BehaviorTree/BehaviorTree.CPP #123: Ports in Condition Node

Sarath18, 2019-10-27:
I am unable to find a way to use input ports in the condition node. The documentation has examples on the same, only for action nodes and not condition nodes. The only workaround I can think of now is to use convert the condition node to the action node. There must be a proper solution to this as the documentation also states:
> ConditionNodes are equivalent to ActionNodes, but they are always atomic and synchronous, i.e. they must not return RUNNING.

My behaviour tree structure is as follows:
```XML
<root main_tree_to_execute="BehaviorTree">
    <!-- ////////// -->
    <BehaviorTree ID="BehaviorTree">
        <Sequence>
            <Condition ID="WaypointReached" threshold_distance="2"/>
            <Action ID="PublishNextWaypoint"/>
        </Sequence>
    </BehaviorTree>
    <!-- ////////// -->
    <TreeNodesModel>
        <Action ID="PublishNextWaypoint"/>
        <Condition ID="WaypointReached">
            <input_port name="threshold_distance" default="2">Threshold distance to waypoint</input_port>
        </Condition>
    </TreeNodesModel>
    <!-- ////////// -->
</root>
```
I want to use threshold_distance input port in the WaypointReached condition.

facontidavide, 2019-10-27:
It you share the c++ code with me, I can have a look

Sarath18, 2019-10-27:
Sure. Here is the [link](https://gist.github.com/Sarath18/2169ff5aadb000a9d0b5fa6333cc377c) to the file. This is without the input port in condition node.

Sarath18, 2019-10-27:
I was using the simple condition node instead of a condition node. I updated my code to use Condition Node instead. This is the error that I am facing now

```
/opt/ros/melodic/include/behaviortree_cpp/bt_factory.h:172:9: error: static assertion failed: [registerBuilder]: Some methods are pure virtual. Did you override the methods tick() and halt()?
         static_assert(!std::is_abstract<T>::value,
         ^~~~~~~~~~~~~
/opt/ros/melodic/include/behaviortree_cpp/bt_factory.h:182:9: error: static assertion failed: [registerBuilder]: the registered class must have at least one of these two constructors:   (const std::string&, const NodeParameters&) or (const std::string&).
         static_assert(default_constructable || param_constructable,
         ^~~~~~~~~~~~~
```

facontidavide, 2019-10-27:
There are multiple error in your code:

- You forgot the word **public** in your class. Everything is private
- If  **threshold_distance** is a numerical value, why are you using std::string?
- DO NOT RETURN RUNNING in as ConditionNode
- Now that you converted WaypointNodes into a class, use  **factory.registerNodeType<>**

Attached there is the corrected code (stripped):

```c++
class WaypointReached : public BT::ConditionNode
{
  public:
  WaypointReached(const std::string& name,
                  const NodeConfiguration& config) :
    BT::ConditionNode(name, config) {}

  static PortsList providedPorts()
  {
    return { InputPort<double>("threshold_distance") };
  }

  BT::NodeStatus tick() override {

    Optional<double> msg = getInput<double>("threshold_distance");

    if(msg) {
      std::cout << msg.value() << std::endl;
      return BT::NodeStatus::SUCCESS;
    }
    else{
      std::cerr << "missing parameter [threshold_distance]" << std::endl;
      return BT::NodeStatus::FAILURE;
    }
  }
};

int main(int argc, char** argv) 
{
  BehaviorTreeFactory factory;
  factory.registerNodeType<WaypointReached>("WaypointReached");
  return 0;
}
```

closed 2019-10-27 by facontidavide