BehaviorTree/BehaviorTree.CPP #125: Updating to v3 using ros apt results in collision

naveedhd, 2019-10-29:
On an already installed `behaviortree-cpp` (using apt install ros-distro-behaviourtree-cpp), installing `behaviortree-cpp-v3` results in following error:

```
sudo apt install ros-dashing-behaviortree-cpp-v3
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  ros-dashing-behaviortree-cpp-v3
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/228 kB of archives.
After this operation, 1,082 kB of additional disk space will be used.
(Reading database ... 310384 files and directories currently installed.)
Preparing to unpack .../ros-dashing-behaviortree-cpp-v3_3.0.9-1bionic.20191012.082854_amd64.deb ...
Unpacking ros-dashing-behaviortree-cpp-v3 (3.0.9-1bionic.20191012.082854) ...
dpkg: error processing archive /var/cache/apt/archives/ros-dashing-behaviortree-cpp-v3_3.0.9-1bionic.20191012.082854_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/dashing/bin/bt_log_cat', which is also in package ros-dashing-behaviortree-cpp 2.5.2-1bionic.20191012.082847
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/ros-dashing-behaviortree-cpp-v3_3.0.9-1bionic.20191012.082854_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

eventually just `apt removed` and installed v3 but it does also remove dependent packages.

facontidavide, 2019-10-30:
this is unfortunately a know issue.

A partial solution is to rename the include folder, breaking anyone's code.

For the file in the bin directory, it is more complicated, unfortunately.

facontidavide, 2019-10-30:
3.1.0 fix this

closed 2019-10-30 by facontidavide