BehaviorTree/BehaviorTree.CPP #129: Should SimpleConditionNode change its status to running?

umhan35, 2019-11-02:
The simple action node change its status to running from idle: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/condition_node.cpp#L31

But the simple condition just runs the ticker function and does not change its status to running from idle : https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/action_node.cpp#L39

Is it a bug? Or it's designed this way.

facontidavide, 2019-11-03:
A condition is never running, because it is an atomic operation. I don't think it is a bug. Does this affect you in any way?

facontidavide, 2019-11-03:
Remember that status is changed also in executeTick

umhan35, 2019-11-03:
I see. So condition is only for things like "CheckBattery" or "IsDoorOpen" or "IsDoorOpen"?

Is it only conceptually different from actions? But in the code, they are technically the same thing?

I need a condition to be in running status because I want the user know it is checking if a door is open (which can take long time to execute for computer vision)

umhan35, 2019-11-03:
We probably need a Ticking status? Otherwise, is there anyway we can know which node is running while a condition node is ticking?

facontidavide, 2019-11-04:
No, what you need are reactive nodes.

facontidavide, 2019-11-04:
If you give me more detail about the behavior you want to achieve, I may help you figuring out the correct design pattenr.

Byt a ticking or running Condition makes no sense in the current semantic.

umhan35, 2019-11-19:
We are working on robot explanations. To answer what you are doing, we need to know which node is currently executing. Even though condition nodes takes relatively short amount of time to execute, we still want the robot know when the question is asked.

facontidavide, 2020-03-19:
> I see. So condition is only for things like "CheckBattery" or "IsDoorOpen" or "IsDoorOpen"?
> Is it only conceptually different from actions? But in the code, they are technically the same thing?

From a "semantic" point of view, the only difference between a Condition and an Action is that it a Condition must never return RUNNING and is considered "atomic", no matter how long it takes.

A part from that, there is no difference. In your loggers, you should still be able to observe the transition from IDLE to SUCCESS, therefore you should be able to observe the execution of a Condition.

If that is not the case, let me know.

closed 2020-03-19 by facontidavide