BehaviorTree/BehaviorTree.CPP #130: Get full plugin library name

orduno, 2019-11-04:
Calling `BTFactory::registerFromPlugin` with the library name results in a *library not found* error, perhaps because the os-dependent filename is not generated.

Getting the library filename is already implemented on `BT::SharedLibrary::getOSName`.

facontidavide, 2019-11-04:
I think that I understand 8and agree) with your suggestion.

But to be sure that we are on the same page, can you please give me more details about how  BTFactory::registerFromPlugin should work in your opinion?

facontidavide, 2020-03-19:
In theory, this method expects the full path, I will not change the API. 

In your case, I suggest doing something like this:

```C++
std::string directory_path("/home/orduno/");
std::string library_filename =  BT::SharedLibrary::getOSName("my_bt_library");
factory.registerFromPlugin( directory_path + library_filename );
```

closed 2020-03-19 by facontidavide