BehaviorTree/BehaviorTree.CPP #131: Not able to compile with ament

peterpena, 2019-11-07:
ros2_devel branch can be compiled with cmake but when I try to compile this with ament, I get an error

`CMake Error at sample_nodes/CMakeLists.txt:17 (set_target_properties):
  set_target_properties called with incorrect number of arguments.`

When I change the CMAKE arguments to `BEHAVIOR_TREE_BIN_DESTINATION` , ament is able to compile again.

I know this was not a problem with version 3.0.9 in branch ros2_dev

facontidavide, 2019-11-07:
I have made a commit that might help. Can you check it out?

peterpena, 2019-11-07:
The master branch compiles with a warning:

```
colcon build --symlink-install --packages-select behaviortree_cpp_v3
Starting >>> behaviortree_cpp_v3
[Processing: behaviortree_cpp_v3]                             
--- stderr: behaviortree_cpp_v3                                  
CMake Warning:
  Manually-specified variables were not used by the project:

    CATKIN_INSTALL_INTO_PREFIX_ROOT

---
Finished <<< behaviortree_cpp_v3 [53.5s]

```

To compile this with ros2, do we not need to make these changes for the ros2_devel branch?

peterpena, 2019-11-07:
The ros2_devel branch compiles now. Thank you!

closed 2019-11-07 by peterpena