BehaviorTree/BehaviorTree.CPP #133: How to correctly reload a behavior tree on the fly?

maxlein, 2019-11-20:
Is there a simple way to gracefully stop the running tree and reload another one?  

Right now I do it pretty ugly, just calling  
`tree = std::make_unique<BT::Tree>( factory.createTreeFromText( currentTreeXml, blackboard ) );`  and letting the destructor do all the work.  

With the tree itself I have no problem, but the PublisherZMQ Logger segfaults at times in its callback method. I think there are some messages left to be sent, but tree is already gone or something like that.  

Tree runs at 10Hz.

facontidavide, 2019-11-21:
Actually I don't think that what you are doing is ugly. The destructor was specifically designed to do this in a clean way.

On the other hand, I need to fix the problem with the Publisher....

facontidavide, 2020-03-19:
Have you tried somethink like this:

``` C++
//When you load the tree the first time

auto tree = std::make_unique<BT::Tree>( factory.createTreeFromText( currentTreeXml, blackboard ) );
auto zmq_publisher = std::make_unique<PublisherZMQ<(*tree);

//When you reload
zmq_publisher.reset(); // to do a clean stop first
tree = std::make_unique<BT::Tree>( factory.createTreeFromText( currentTreeXml, blackboard ) );
 zmq_publisher = std::make_unique<PublisherZMQ<(*tree);
```

maxlein, 2020-03-20:
Will try that, thanks.

Nikpec, 2020-03-20:
Thank you very much for the hint to the reset function of PublisherZMQ. Your code above with the reset function works for me.

facontidavide, 2020-03-20:
I am happy to hear that. Arguably, the code is a little bit ugly, 

I think I should consider an update ot the API to make it nicer.

closed 2020-03-20 by facontidavide