BehaviorTree/BehaviorTree.CPP #137: Re-merge #53

facontidavide, 2019-12-09:
Lost in another branch?

HansRobo, 2019-12-11:
This problem is a little complex.

The merge request #53 is merged into "ver_3"  upstream branch.
But after that, upstream "ver_3" branch never merged into master branch and deleted upstream "ver_3" branch.

The following is my imagination, the detailed timeline is below.

1. there is "ver_3" branch(upstream)
2. you made some commits on "ver_3" branch(upstream)
3. you merged "ver_3" (upstream) into local "ver_3" (2019/2/27 20:58)
4. you made some commits on "ver_3" (local) branch 
6. Pull Request #53 was merged into "ver_3"(upstream) (2019/2/28 01:51)
5. you merged "ver_3"(local) into "master" branch (2019/2/28 01:59)
7. someone deleted "ver_3"(upstream) branch

From the timeline, it is clear that "ver_3"(upstream) branch was deleted with content of #53.

[image]

HansRobo, 2020-03-02:
Thank you for accepting #138 
And this issue should be closed

closed 2020-03-02 by facontidavide