BehaviorTree/BehaviorTree.CPP #140: zmq publisher cat't use parallelly

HansRobo, 2019-12-17:
zmq publisher can't use parallelly, because tcp port is fixed as below.
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/2a859e68f40f0b86c910e2f76053a3e9b5a09469/src/loggers/bt_zmq_publisher.cpp#L43-L44

If you try to use zmq publishers in parallel, you will get an error on the second and subsequent publishers.
~~~
terminate called after throwing an instance of 'zmq::error_t'
  what():  Address already in use
~~~

This problem makes me inconvenient when debugging multi-robots for example, RoboCup Soccer SSL.

Are there any plans to expand it to allow flexible port specification?

facontidavide, 2020-03-19:
Fixed, at least on this library side. I need to address this in Groot now.

Thanks for reporting!

Sarath18, 2020-04-16:
Thank you for the update!
I have a problem. I can't create multiple instances of PublisherZMQ with different server and publisher port configuration even on different threads in the same program. This works perfectly if they are on separate executables or instances of application with configured ports. What is the workaround to publish status of multiple trees from the same program?

SteveMacenski, 2021-03-24:
I've done this in Nav2 via making the ZMQ publisher a static variable shared across all instances, so far the only workable solution I've found other than making it an input reference.

closed 2022-11-19 by facontidavide