BehaviorTree/BehaviorTree.CPP #143: CoroActionNode

benmoggee, 2020-01-10:
Hi,

I get this error when using CoroActionNode. I run on MACOS the latest.

 ./t09_async_actions_coroutines
action_A: Started. Send Request to server.
action_A: Waiting Reply...
t09_async_actions_coroutines(2246,0x10c055dc0) malloc: Incorrect checksum for freed object 0x7fa299c03e48: probably modified after being freed.
Corrupt value: 0x7fff6672b093
t09_async_actions_coroutines(2246,0x10c055dc0) malloc: *** set a breakpoint in malloc_error_break to debug
zsh: abort      ./t09_async_actions_coroutines

facontidavide, 2020-03-18:
I know it doesn't help but... it works on my machine :(

I don't know how to reproduce this error

facontidavide, 2020-03-22:
I refactored the entire CoroAction implementation. 

Now, it will require Boost 1.59+

Note that in most cases you may replace CoroAction with something more straightforward like **StatefulActionNode**

closed 2020-03-22 by facontidavide

benmoggee, 2020-03-23:
Thanks. I got it to work by doing what was recommended in this issue. I will look at the new implementation a little later.

[image]