BehaviorTree/BehaviorTree.CPP #144: CMake warning of not finding the library

maxlein, 2020-01-29:
What's about this warning:  

```
-- Found behaviortree_cpp_v3: 3.1.1 (/opt/ros/eloquent/share/behaviortree_cpp_v3/cmake)
CMake Warning at /opt/ros/eloquent/share/behaviortree_cpp_v3/cmake/ament_cmake_export_libraries-extras.cmake:116 (message):
  Package 'behaviortree_cpp_v3' exports library 'behaviortree_cpp_v3' which
  couldn't be found
```  

I have installed the ros2 eloquent version and I see this warning when building the nav2_behavior_tree and a custom node of my own.  

Interesting thing is, when building with colcon, I have no error.  
But building with make in clion I get linker errors to the BT lib.  
The warning is received in both ways.  

Tried deleting build folder / cmake cache but no success.  

Can you give sme hints where to look for, seems to be the installed lib.

facontidavide, 2020-03-19:
no idea

facontidavide, 2020-03-22:
I just pushed a change in the cmake that should fix this issue in the next releases (3.2+)

facontidavide, 2020-03-22:
I will tentatively close this. Please download the latest version and tell me if the problem persists

closed 2020-03-22 by facontidavide

maxlein, 2020-03-23:
You may have the commit where you changed this?