BehaviorTree/BehaviorTree.CPP #146: How to reset CoroAction Node to a clean state.

Jconn, 2020-02-11:


facontidavide, 2020-03-18:
The answer is not what you may expect...

Do not use CoroAction!

My sad story is that I nerited the **AsyncAction** from version 1.X. I finally realized that it is almost impossible for me to avoid race conditions and to correctly instruct people how to implement this to make it really asynchronous.

I tried to solve this problem with the coroutines. This solved the problem of race conditions, but, as you noticed, it is still unclear how to correctly clean up your state.

Nowadays (I need to reflect this in the documentation, I think the best model is **StatefulAction** 

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/action_node.h#L158

Where it is much cleaner how to implement an asynchronous node with a cleanup.

I need to dif into this to see if I can find a solution.

facontidavide, 2020-03-19:
Anyway, to close this issue.

I recently made the decision to forbid my used from using setStatus() https://github.com/BehaviorTree/BehaviorTree.CPP/commit/4f30e3a42ff0c657eb5711d7ae94318fde93bd60

The only correct way to "stop" a node is not by setting its state to IDLE (you shouldn't, only the parent node should), by rather calling halt()

facontidavide, 2020-03-19:
I finally understoof the problem:

> Even though we set the CoroActionNode to status idle, that doesn’t reset the state of:
> _p->coro or _p->pending_destroy in that object.

I have **never** expected multiple threads to  run the` tree.root_node->executeTick()`

In fact`CoroAction` is already robust to the fact that a different thread may calls `halt()`... I forgot that I fixed that in the past!

But it will fail when multiple threads call `executeTick().`

I don't have a "solution". Either:

- don't use CoroAction. looking at the [source code here](https://github.com/ros-planning/navigation2/blob/master/nav2_behavior_tree/include/nav2_behavior_tree/bt_action_node.hpp), the change should be pretty easy to do.

- be sure that there is only one thread executing the thread. I would love to do some code review with you to understand why you are spawning different threads every time.

facontidavide, 2020-03-22:
I am happy to say that this problem is solved.

I replace the old coroutine library with boost::coroutine2, that solves the problem

https://github.com/BehaviorTree/BehaviorTree.CPP/commit/02330ac036621f7ad3c169abde88525ff5463463

closed 2020-03-22 by facontidavide