BehaviorTree/BehaviorTree.CPP #147: Citation for academic paper

SteveMacenski, 2020-02-11:
Hi,

I don't suppose you have a paper outlining this work i can cite in an academic paper. I would love to give credit where credit is due.

S

miccol, 2020-02-12:
Not sure if @facontidavide has a paper related to this specific repo. 
Sometimes people create a BibTeX entry for GitHub repos as well. An attempt for this repo could be:

```
@misc{BTCpp
  author = {Davide Faconti, Michele Colledanchise},
  title = {{BehaviorTree.CPP}},
  version      = {3.0},
  year = {2019},
  publisher = {GitHub},
  journal = {GitHub repository},
  url    = {https://github.com/BehaviorTree/BehaviorTree.CPP}
}
```

However, it reports only the main authors (I also acknowledge that I haven't contributed to the repo in a while) other people improved it with PRs etc.

facontidavide, 2020-02-12:
I agree with Michele. In addition, this is a well structured report that explains the "contribution" of this library compared with other implementations (and models) in the literature.

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/MOOD2Be_final_report.pdf

closed 2020-02-12 by facontidavide