BehaviorTree/BehaviorTree.CPP #149: Structure issure with log file

ipa-rwu, 2020-03-02:
Hello,
I try to get log file of behavior tree. I am using code ```

```
    StdCoutLogger logger_cout(tree);

    // This logger saves state changes on file
    FileLogger logger_file(tree, "bt_trace.fbl");

    // This logger stores the execution time of each node
    MinitraceLogger logger_minitrace(tree, "bt_trace.json");
```
But I checked `bt_trace.json` file. It seems wrong.  For example:

```
{"traceEvents":[
{"cat":"","pid":0,"tid":-658726592,"ts":9244,"ph":"B","name":"Pick_hold_place","args":{}},
{"cat":"@","pid":0,"tid":-658726592,"ts":9574,"ph":"B","name":"Back home","args":{}},
{"cat":"á*€oç¿","pid":0,"tid":-658726592,"ts":9772,"ph":"I","name":"BTIsObjContainer","args":{}},
{"cat":"","pid":0,"tid":-658726592,"ts":9828,"ph":"E","name":"Back home","args":{}},
{"cat":"@","pid":0,"tid":-658726592,"ts":9853,"ph":"B","name":"Place","args":{}},
{"cat":"á*€oç¿","pid":0,"tid":-658726592,"ts":9893,"ph":"I","name":"BTIsHoldObj","args":{}},
{"cat":"","pid":0,"tid":-658726592,"ts":9935,"ph":"E","name":"Place","args":{}},
{"cat":"@","pid":0,"tid":-658726592,"ts":10146,"ph":"B","name":"Pick","args":{}},
{"cat":"á*€oç¿","pid":0,"tid":-658726592,"ts":10204,"ph":"I","name":"BTIsObjPose","args":{}},
{"cat":"","pid":0,"tid":-658726592,"ts":10246,"ph":"E","name":"Pick","args":{}},
{"cat":"@","pid":0,"tid":-658726592,"ts":10267,"ph":"B","name":"Sequence","args":{}},
{"cat":"9âÙíkÁV?","pid":0,"tid":-658726592,"ts":10436,"ph":"B","name":"BTCameraFindTarget","args":{}},
{"cat":"á*€oç¿","pid":0,"tid":-658726592,"ts":832216,"ph":"I","name":"SetBlackboard","args":{}},
{"cat":"h¡ÿ>","pid":0,"tid":-1778403584,"ts":832225,"ph":"E","name":"BTCameraFindTarget","args":{}},
{"cat":"á*€oç¿","pid":0,"tid":-658726592,"ts":832301,"ph":"I","name":"SetBlackboard","args":{}},
```

In terminal it loos like this:

```
[1583186744.607]: Sequence                  RUNNING -> SUCCESS
[1583186744.607]: Sequence                  SUCCESS -> IDLE
[1583186744.607]: Sequence                  SUCCESS -> IDLE
[1583186744.607]: SequenceStar              RUNNING -> SUCCESS
[1583186744.607]: Execution                 RUNNING -> SUCCESS
[1583186744.607]: SequenceStar              SUCCESS -> IDLE
[1583186744.607]: SetBlackboard             SUCCESS -> IDLE
[1583186744.607]: Place_subtree             SUCCESS -> IDLE
[1583186744.607]: Execution                 SUCCESS -> IDLE
[1583186744.607]: place_loop                RUNNING -> SUCCESS
[1583186744.607]: BTIsHoldObj               SUCCESS -> IDLE
[1583186744.607]: place_loop                SUCCESS -> IDLE
[1583186744.607]: Place                     RUNNING -> SUCCESS
[1583186744.607]: Back home                 FAILURE -> IDLE
[1583186744.607]: Place                     SUCCESS -> IDLE
[1583186744.607]: Pick_hold_place           RUNNING -> SUCCESS
[1583186744.607]: Pick_hold_place           SUCCESS -> RUNNING
[1583186744.607]: Back home                 IDLE -> RUNNING
[1583186744.607]: BTIsObjContainer          IDLE -> FAILURE
[1583186744.607]: BTIsObjContainer          FAILURE -> IDLE
[1583186744.607]: Back home                 RUNNING -> FAILURE
[1583186744.607]: Place                     IDLE -> RUNNING
[1583186744.607]: BTIsHoldObj               IDLE -> SUCCESS
[1583186744.607]: place_loop                IDLE -> RUNNING
[1583186744.607]: SetBlackboard             IDLE -> SUCCESS
[1583186744.607]: Place_subtree             IDLE -> RUNNING
[1583186744.607]: Fallback                  IDLE -> RUNNING
[1583186744.607]: Check back                IDLE -> RUNNING
[1583186744.607]: Placed?                   IDLE -> RUNNING
[1583186744.607]: SetBlackboard             IDLE -> SUCCESS
[1583186744.654]: BTCheckCondition          IDLE -> SUCCESS
[1583186744.655]: SetBlackboard             IDLE -> SUCCESS
[1583186744.655]: SetBlackboard             SUCCESS -> IDLE
```

Is anyone have same issues?

facontidavide, 2020-03-02:
Why do you think it looks wrong? it is not meant to be human readable, but visualized with Chrome.

See here: https://github.com/hrydgard/minitrace

ipa-rwu, 2020-03-03:
I try to load `bt_trace.json` in `chrome://tracing/` . But i got 
[image]. Did I do anything wrong?

facontidavide, 2020-03-03:
I will check it out

facontidavide, 2020-03-06:
Fixed! Thanks for reporting

closed 2020-03-06 by facontidavide