BehaviorTree/BehaviorTree.CPP #151: Question: How does this compare to cpp-taskflow

Levi-Armstrong, 2020-03-05:
We are currently looking for a library to develop a process planning framework and I believe I have narrowed it down to this library and [cpp-taskflow](https://github.com/cpp-taskflow/cpp-taskflow). I have used cpp-taskflow but have not had the opportunity to use BehaviorTreeCPP but I am very interested and wanted to get the developers input. Would these be competing solutions or would the case be to leverage cpp-taskflow within a behavior tree node?

miccol, 2020-03-06:
Hi,
Thank you for the question.
Do you have a document (e.g. technical report, paper etc) that describes the semantic of the taskflows?

Thanks

facontidavide, 2020-03-06:
I have been looking in the past to cpp-taskflow myself to implement parallel computation. Seriously, we are talking about two different things here that has no relationship to each other... Almost 😜

Levi-Armstrong, 2020-03-06:
Maybe it is better to explain what I am trying to implement and see if this a right fit or if I should be leveraging both. In the image below it is an example of a process path for say a sanding process where there are several approaches for going about generating the robot trajectory. 

One of our approaches, is we first solver for the process paths using a sample based planner called Descartes followed by an optimization planner called TrajOpt for smoothing and other constraints. All of these could be computed in parallel.

Second we then plan for the approach and departures using the same process described above which also can be computed in parallel.

Third we then plan for the transitions which depending on the process it is either a cartesian path or a freespace path. In the case of cartesian it is the same process using Descartes then TrajOpt but if it is freespace we use OMPL Planner followed by TrajOpt to clean things up. These also can be planned in parallel. 

The final step is to plan the transition from start and to end using OMPL followed by TrajOpt. These both can be planned in parallel. 

With the cpp-taskflow I could layout this whole process and execute the task, but I would like it to make it more sophisticated for situations where one of the planners. Which is where I thought BehaviorTreeCPP fits where I could define a behavior for the each of the steps to handle when planners fail for example and analyze the failure and make some parameter changes and replan etc.. Also it would be nice to have a high level behavior tree that handles the overall process which make decisions when different sections of the process planning fails or succeeds. 

I am trying to figure out could the same parallel compute be accomplished within BehaviorTreeCPP, if not should I leverage cpp-taskflow within a node? Our is BehaviorTreeCPP not a good fit for this?

[image]

facontidavide, 2020-03-06:
Indeed, cpp_taskflow is the best solution when you have a pipeline/graph of parallel tasks that is "static".

BehaviorTRee is potentially able to do things in Parallel (not so well tested, but implemented) and to add some additional logic about "if these conditions are met, do this or do something else.

I need to think about this and try to build with BT.CPP a dummy implementation to see if it is a good fit...

miccol, 2020-03-09:
> Maybe it is better to explain what I am trying to implement and see if this a right fit or if I should be leveraging both. In the image below it is an example of a process path for say a sanding [...]

Could you provide us a graph of the `cpp-taskflow` for that?

Thanks

facontidavide, 2020-03-09:
If I am honest with you, the best library to solve your problem is this one: https://github.com/google/marl

If will give you a very easy to use API to do parallel computation (easier than BehaviorTree) and you can add your logic with plain old code.

closed 2020-03-19 by facontidavide