BehaviorTree/BehaviorTree.CPP #153: Default input port value

shivaang12, 2020-03-11:
@facontidavide I am using default value for input ports but couldn't able to make it work. Can you point me to unit test which test this feature? (as you mentioned on #56 about unit test.)

Thanks!

leewo0, 2020-03-12:
@shivaang12 ”default_value“ of "InputPort" and "OutputPort" is misunderstanding. After i checked the source code, i found the "default_value" is used to set the port remapping value if an inputport's remapping value is missed. And there might be a bug, as the source code in xml_parsing.cpp shows:
`        // use default value if available for empty ports. Only inputs
        for (const auto& port_it: manifest.ports)
        {
            const std::string& port_name =  port_it.first;
            const PortInfo& port_info = port_it.second;

            auto direction = port_info.direction();
            if( direction != PortDirection::INPUT &&
                config.input_ports.count(port_name) == 0 &&
                port_info.defaultValue().empty() == false)
            {
                config.input_ports.insert( { port_name, port_info.defaultValue() } );
            }
        }` 
`direction != PortDirection::INPUT` should be `direction == PortDirection::INPUT`.
Anyway, i also want to find out the way set a default value for a blackboard value. (I have to initialize them within extra nodes.)

facontidavide, 2020-03-12:
it is most probably a bug. I need to look into this... soonish

facontidavide, 2020-03-18:
Thanks for insisting ;)

it was indeed a bug. Fixed.

You can find an example here: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/gtest_ports.cpp

closed 2020-03-18 by facontidavide

SteveMacenski, 2020-03-19:
@facontidavide when can we expect to see this in a release?

facontidavide, 2020-03-19:
I am trying to fix as many open issues as possible, including some that may affect Navigation2, to release it.