BehaviorTree/BehaviorTree.CPP #165: Provide (or document) `find_package` for cmake

buddha314, 2020-03-22:
If a `find_package` is provided in the source, i do not see it.  Please provide documentation and the name of the library.

facontidavide, 2020-03-22:
Just updated the readme. Have a look

closed 2020-03-22 by facontidavide

buddha314, 2020-03-22:
Thank you so much!  This might not be a BT problem, but the current error is

```
/usr/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /mnt/c/Users/b/CLionProjects/beetree
-- Configuring done
CMake Error at CMakeLists.txt:8 (add_executable):
  Target "beetree" links to target "BT::behaviortree_cpp_v3" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?
```

With CMakeLists.txt

```
cmake_minimum_required(VERSION 3.5)
project(beetree)

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
find_package(BehaviorTreeV3)

add_executable(${PROJECT_NAME} "main.cpp")

target_link_libraries(${CMAKE_PROJECT_NAME} BT::behaviortree_cpp_v3)
```

if you have thoughts, i would appreciate it.

buddha314, 2020-03-22:
Ah, so an obvious one, the `BT::` in `target_link_libraries` should be removed.  Thank you so much and let me suggest you update the README.

facontidavide, 2020-03-22:
That is weird because on my computer it does work...

Can you please check the file /usr/local/lib/BehaviorTreeV3/cmake/BehaviorTreeV3Config.cmake

at line 48, that should be

           add_library(BT::behaviortree_cpp_v3 SHARED IMPORTED)

buddha314, 2020-03-23:
```
 44 # Compute the installation prefix relative to this file.
 45 get_filename_component(_IMPORT_PREFIX "${CMAKE_CURRENT_LIST_FILE}" PATH)
 46 get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
 47 get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
 48 get_filename_component(_IMPORT_PREFIX "${_IMPORT_PREFIX}" PATH)
 49 if(_IMPORT_PREFIX STREQUAL "/")
 50   set(_IMPORT_PREFIX "")
 51 endif()
 52
 53 # Create imported target BehaviorTreeV3::behaviortree_cpp_v3
 54 add_library(BehaviorTreeV3::behaviortree_cpp_v3 SHARED IMPORTED)
 55
 56 set_target_properties(BehaviorTreeV3::behaviortree_cpp_v3 PROPERTIES
 57   INTERFACE_COMPILE_DEFINITIONS "ZMQ_FOUND"
 58   INTERFACE_INCLUDE_DIRECTORIES "${_IMPORT_PREFIX}/include;/usr/include"
 59   INTERFACE_LINK_LIBRARIES "-lpthread;dl;/usr/lib/x86_64-linux-gnu/libzmq.so"
 60 )
```

This is on Ubuntu as a Windows Subsystem for Linux if that helps.

facontidavide, 2020-03-23:
This is an "old" (few hours old) version .

If you recompile and install the very last version on master, the correct prefix will be  BT::, so the README is correct as far as I know.

buddha314, 2020-03-23:
I recompiled with a fresh pull just now.  The BehaviorTreeV3Config file did not seem to change, it looks like that file wasn't over-written at all.

To be clear, I was able to compile yesterday, but my CMakeLists does not include `BT::` So I can continue my noodling.  I'm happy to try new builds at your request, but I am not blocked at the moment.