BehaviorTree/BehaviorTree.CPP #170: Retry until success sequence of subtree

capox95, 2020-04-03:
Hi, I am working on the following Behavior Tree:
[image]

BIN_LOCALIZATION and all the others children are SubTrees.

I want to achieve the following behavior:
the SubTree composed by INSERTION_CLOTH, INSERTION_CHECK and RECOVERY_GRASP should loop if RECOVERY_GRASP returns with status of success. Instead, if INSERTION_CHECK returns with success, the execution should restart from the beginning of the behavior tree (this works!).

I'am able to achieve this result when I work on that subtree alone since I can control the condition of ticking from the main file (where the nodes are registered).

What is the best way to achieve this behavior with the tools available in this library?
I am tinking of a sequence node with a condition of "retry until success". I have read that there exist decorators like this but they can only have a single child.

[image]

I hope the problem is clear,
Alessio

closed 2020-04-03 by capox95

facontidavide, 2020-04-03:
Yes, RetryUntilSuccesful is exactly what you need.

But the example you have here is not correct. To understand better and be able to give you a suggestion: can Recovery Grasp fail? if Yes, what do you want to do if it fails?

capox95, 2020-04-03:
Hi, thanks for the reply.
To answer you question, for the time being I am focusing on the "nominal" behavior, later on I will address the possible points of failure. So Recovery Grasp is assumed to always succeed. 

I spent the morning thinking on a possible solution, so I opened this issue. After I realized that RetryUntilSuccesful can only have one child but the child can be a subtree. My bad!