BehaviorTree/BehaviorTree.CPP #172: Out-of-bounds crashes with the ZeroMQ publisher.

unvestigate, 2020-04-10:
I recently tried out the ZeroMQ publisher and the monitor in Groot. When trying to monitor anything but the very simplest of BTs I get an out-of-bounds error on the Groot side. However, after some brief investigations I believe the bug is in BT.CPP, not Groot, which is why I am raising the issue here.

The crash I get is in sidepanel_monitor.cpp, line 56, ie. on this line:

`AbstractTreeNode* node = _loaded_tree.node( index );`

The index is read from the ZMQ message and it is out-of-bounds. The bug seems to be the fact that  the publisher writes the "UID" of each node which is a running number, in PublisherZMQ::createStatusBuffer(), to the message. However, Groot treats it as an index. Since the UID is based off of a static value in getUID() in tree_node.cpp it means that if you happen to have two BTs instantiated in the same process, you have a 50% chance of crashing, depending on which BT was instatiated first and which one you are trying to monitor. If I disable every tree except one and try to monitor it, it is seems to work which supports my theory.

However, I also get an out-of-bounds crash in the same place if I use SubTreeWrappers in my one-and-only BT which suggests that sub-trees also cause this kind of issue.

Seeing as a video games easily can load 20-30 BTs for a single level this makes the monitoring effectively broken.

facontidavide, 2020-04-10:
thanks for reporting, I will look at this issue

facontidavide, 2022-11-19:
fixed

closed 2022-11-19 by facontidavide