BehaviorTree/BehaviorTree.CPP #174: How should we implement user input?

AlexisTM, 2020-04-15:
Hi, 

We are revamping our robot behavior and we are looking into BehaviorTrees but we don't understand how to implement the user input.

We have multiple exclusive actions triggered by user inputs such as "AutoCalibrate", "GeneratePlan", "DoPlan", "GoToChargingStation" or "Stop". All these actions can be plugged together to get a meaningful behavior but should also be able to do them each separately and able to be stopped by the "Stop" user input. 

How do we do that "Subtree selection"? As I come from a state machine, I think "I would need a new `Selector` ControlNode" to handle state-like transitions.

facontidavide, 2020-04-15:
A long time ago I implemented a **ManualSelector** using ncurses library.

In that way, the user could manually decide which child should be executed.

It is somehow in my todolist to implement that again and make it available in the main library.

Nut to answer your question: you need to implement your own ControlNode.

facontidavide, 2020-04-16:
I created the **ManualSelector** node. Please have a look at tutorial 12 in the **examples** folder.

closed 2020-04-16 by facontidavide

AlexisTM, 2020-04-17:
Thank you! I will do

facontidavide, 2020-04-17:
Note that ManualSelector will show an interactive menu everytime it is executed.

If I understand correctly your original email, you might want to use always the same value, passed **only once** by the user.

A solution would be to ask for the action to do once with the ManualSelector, write the result in the BlackBoard and then use the new ControlNode Switch:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/controls/switch_node.h

facontidavide, 2020-04-17:
Have a look to the latest change. It might be useful for you.

https://github.com/BehaviorTree/BehaviorTree.CPP/commit/4a8cf33f3c20190542fe1ff053202dee90eb1c23

AlexisTM, 2020-04-17:
Actually, it is an input coming from an external web interface and the input should be used once only, after finished or receiving "Stop" (itself an input), we wait for another input.