BehaviorTree/BehaviorTree.CPP #183: Getting Blackboard variable name

Brudasoe, 2020-04-27:
Hi, if I put a Blackboard variable in an input port of my node, is there a way to access it's name in cpp code?

facontidavide, 2020-04-27:
I am not sure I follow you can you give me an example maybe, based on the tutorials?

If you mean the name of the entry in the Blackboard, no, currently you can't.
What is the use case?

facontidavide, 2020-04-27:
Having a quick look at the code. It is easy to implement, but just curious about the use case.

Brudasoe, 2020-04-28:
I guess it's mainly for logging purposes, for example if I had a custom node that compares different integers stored in Blackboard values, I would like to log "RobotDistance is lower than ThresholdDistance". 
I can do that with a specific case, but if I wanted to reuse that node in other place, to check for example "GripperWidth is lower than ThresholdGripperWidth", then I'd have to duplicate the code. If I get access to those variable names, then I can give them to the logging function.
I hope this clears it up a bit

facontidavide, 2020-04-28:
I am still missing the point.

any port is accessed by name. You do have that name. The actual content of the port might be a hard-coded value or the name of the blackboard location.

Anyway, considerer this example:

```XML
<SaySomething message={greetings}/>
```

What I can do is to provide a method like

```C++
Optional<StringView> getRawPortValue("message");
```

That will return the string **"{greetings}"**. It is up to you to decide what to do with this information.
You can check if it follows the "blackboard entry pattern" using the static methods:

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/tree_node.h#L143-L145

Is that what you need?

Brudasoe, 2020-04-29:
Yes, 
to better illustrate let's say I have a sequence: 
```
<Sequence>
<SetBlackboard output_key="RobotStatus" value="OK"/>
<SetBlackboard output_key="RobotStatus" value="NO_CONNECTION"/>
<SaySomething     message="{RobotStatus}" />
<SaySomething     message="{NetworkStatus}" />
</Sequence>
```
Inside `SaySomething` or some other custom node I could put a code that would print:
```
std::cout << "Current " << getRawPortValue("message") << " is " << getInput<std::string>("message").value() << std::endl;
```

Which would result in:
"Current {RobotStatus} is OK"
"Current {NetworkStatus} is NO_CONNECTION"

facontidavide, 2020-04-29:
done

closed 2020-04-29 by facontidavide