BehaviorTree/BehaviorTree.CPP #188: How to get a list of the Blackboard keys from a node?

lstelzner, 2020-05-06:
I'm creating a class that inherits from `BT::StatusChangeLogger`  . In this class I want to collect extra data including what is currently in the nodes' Blackboard (keys/values) at the time of the `callback` function. Since my tree has multiple nodes with various input and output ports, I  do not know the key strings from within callback. I've been trying the following but it crashes once it reaches `getKeys()` .
```
void Metrics::callback(BT::Duration timestamp, const BT::TreeNode& node,
                              BT::NodeStatus prev_status, BT::NodeStatus status)
{
 
  BT::Blackboard::Ptr bb(node.config().blackboard);
  if (!bb )
  { 
    ROS_WARN("Getting keys");
    std::vector<BT::StringView> keys = bb->getKeys();
  ROS_WARN("GOT Keys");
  }
}  
```

```
[ WARN] [1588786259.942270451, 1588786258.704000000]: Getting keys
[behavior-1] process has died [pid 127, exit code -11, cmd ...behavior_trees_node __name:=behavior
```
In running in a debugger an exception happens on `out.reserve( storage_.size() );`  https://github.com/BehaviorTree/BehaviorTree.CPP/blob/cc3f0d9fc7c241933b416a279e9897b1f71e95e0/src/blackboard.cpp#L78
Leading me to suspect maybe my method of getting the blackboard is incorrect.
Is what I want to do supported by the API, and if so what is the correct way to do it?
Thanks.

facontidavide, 2020-05-06:
You just found a bug. Thanks for the accurate report. Fixed !

msadowski, 2020-06-01:
@facontidavide I've just tried @lstelzner method with a custom logger callback function like this:

```
void TopicLogger::callback(
  BT::Duration timestamp,
  const BT::TreeNode & node,
  BT::NodeStatus prev_status,
  BT::NodeStatus status)
{
    BT::Blackboard::Ptr bb(node.config().blackboard);
    if (!bb )
    {
      ROS_WARN("Getting keys");
      auto keys = bb->getKeys();
      ROS_WARN("GOT Keys");
    }

  return;
}
```

however I also seem to get a segfault when getKeys() is called.

I'm run this with version 3.5.0, that I've got from debs (ros-melodic-behaviortree-cpp-v3). I've tried changing my tree to something minimal, however the problem was still there. Would you have any thoughts on what could be the issue?

facontidavide, 2020-06-01:
I would like to fix this ASAP. Is there a simple code that I can use to replicate the error?

msadowski, 2020-06-01:
@facontidavide I'll try to create a repo with minimal crashing example in the next hour or so, will keep you posted.

msadowski, 2020-06-01:
I've made a short repo that reproduces the issue for me here: https://github.com/msadowski/bt_logger_test

If you build this with catkin and use the bt.launch file then it should hopefully fail (at least it does for me). Hope this repo helps! 

Also please let me know if I'm doing anything wrong there

facontidavide, 2020-06-01:
There is a typo in your code
```
BT::Blackboard::Ptr bb(node.config().blackboard);
    if (!bb )
    {
      ROS_WARN("Getting keys");
      auto keys = bb->getKeys();
      ROS_WARN("GOT Keys");
    }
```

You should write 
```
if ( bb ) {...}
```

You are accessing a **nullptr** :stuck_out_tongue: 
So, you may ask: why is that null?

I was digging into my own code and realized that when any Node (including my own) has only this kind of constructors

       MyNode(const std::string& name);

But **not** this one

       MyNode(const std::string& name, const BT::NodeConfiguration& config)

That node will not have a pointer to the blackboard, since that pointer is actually a field of **NodeConfiguration**.

So, from my point of view it makes sense because I strongly discourage using the blackboard, ports should be used instead.
So a node without ports don't need the blackboard.

What do you think?

msadowski, 2020-06-02:
Hi Davide,

Thanks for spotting the mistake and for the explanation (I think the `!` might have been the core of this issue in the first place since it's already in Istelzner's code). 

Your approach makes a perfect sense, especially that you discourage using blackboard. I'll make sure that all my nodes have ports and I'll be good! I think you can close this issue back!

Many thanks for the feedback!

closed 2020-06-02 by facontidavide