BehaviorTree/BehaviorTree.CPP #198: Release for Foxy (and probably Noetic, while you're at it)

SteveMacenski, 2020-05-27:
Per http://repo.ros2.org/status_page/blocked_releases_foxy.html?s=6&r=1 this is shockingly one of the only dependencies missing for navigation2 to be released in ROS2. I expect it to take a while to get released due to the mountain of dependencies (with maintainers that are clearly more on the ball than me).

You may also want to release in noetic while you're at it.

facontidavide, 2020-05-27:
done

closed 2020-05-27 by facontidavide

SteveMacenski, 2020-05-27:
Groovy, thanks Davide!

doisyg, 2020-06-03:
Yes, thanks! (for Noetic, not using Foxy yet apart for experimenting, @SteveMacenski, you are going to hear a lot from me the day it happens)

facontidavide, 2020-06-03:
> Groovy, thanks Davide!

@SteveMacenski  do you want me to release for Groovy?   http://wiki.ros.org/groovy :laughing:

SteveMacenski, 2020-06-03:
No promises Jacob will accept that rosdistro PR 😉