BehaviorTree/BehaviorTree.CPP #200: Linker Errors using C++17

SegfaultCreator, 2020-05-29:
Hello together, 
while trying to do all the provided tutorials, i ran into a linking issue when defining Nodes that use Input and Output ports. During build process, the following issues are raised when i add the Code to register the Nodes:

> /usr/local/include/behaviortree_cpp_v3/basic_types.h:267: error: undefined reference to `BT::PortInfo::setDescription(std::basic_string_view<char, std::char_traits<char> >)'

> /usr/local/include/behaviortree_cpp_v3/basic_types.h:115: error: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > BT::convertFromString<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(std::basic_string_view<char, std::char_traits<char> >)'

... (6 in total)

During debugging i moved the files outside our main CMake-project and the build was successful.
It turned out, that the issue can be triggered by adding
`set(CMAKE_CXX_STANDARD 17)` in the CMake File. 

I know that the ReadMe file sets the CMake to C++14, however i would really like to use the tree in my C++17 project. Is there any way to fix this issue in C++17 or are there any Releases in sight compatible with C++17?

Thanks in Advance

facontidavide, 2020-05-29:
Compliment for being issue number 200.

As a special gift, you get a prompt solution to your issue (just pushed)

closed 2020-05-29 by facontidavide

SegfaultCreator, 2020-05-29:
Thanks for the impressively quick Answer :+1: 
A Colleague suggested compiling the Lib with C++17, which also works now. 
Keep up the good work!

facontidavide, 2020-05-29:
yes, the problem was that you could not mix c++14 and c++17 together. Now you can

SegfaultCreator, 2020-06-02:
Supplemental info for my suggestion to use C++17 (for anyone googling this stuff):
Compiling BehaviorTree with C++17 will produce issues when using Groot (uses C++11)

facontidavide, 2020-06-02:
not with the latest changes... Grood submodule to be updated accordingly