BehaviorTree/BehaviorTree.CPP #201: Returning the values from subtree

Lakshadeep, 2020-06-01:
I have a sub-tree defined as follows:

```
<BehaviorTree ID="PeopleDetection">
          <Sequence name="people_detection_left_main">
              <Turn name="turn" turn_angle="{angle}"/>
              <RetryUntilSuccesful num_attempts="10">
                  <CameraUpdateMonitor name="camera_update_monitor" />
              </RetryUntilSuccesful>
              <DetectPeople name="detect_people" detected_people="dp" />
              <SetBlackboard output_key="detected_people" value="{dp}" />
          </Sequence>
</BehaviorTree>
```
So the input port for the subtree is `angle` and output port is `detected_people`.

For the action `DetectPeople`, `detected_people` is the output port. I want to remap the value of this port to the output port of the sub-tree as described above in XML.

However, when I execute this I get runtime error with message `missing port [value]`.

In the example:
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/examples/t06_subtree_port_remapping.cpp

The output port of the sub-tree is manually set. Is it possible to remap the output of some action as the output of the subtree? Does this functionality exist? What is the correct way to implement this?

facontidavide, 2020-06-01:
I see a few weird thing here:

1) If this is the SubTree, you are NOT showing me the parent tree where the remapping is done. Remapping is done on the caller side (parent tree), not the callee (subtree).

2) This is wrong:
```
     <DetectPeople name="detect_people" detected_people="dp" />
     <SetBlackboard output_key="detected_people" value="{dp}" />
```
The correct version is 
```
    <DetectPeople name="detect_people" detected_people="{detected_people}" />  
```
That would also be equivalent to this redundant version that set two BB entries (notice the difference with your code):
```
    <DetectPeople name="detect_people" detected_people="{dp}" />
    <SetBlackboard output_key="detected_people" value="{dp}" />
```

facontidavide, 2020-06-01:
There is a typo in your code
```
BT::Blackboard::Ptr bb(node.config().blackboard);
    if (!bb )
    {
      ROS_WARN("Getting keys");
      auto keys = bb->getKeys();
      ROS_WARN("GOT Keys");
    }
```

You should write 
```
if ( bb ) {...}
```

You are accessing a **nullptr** :stuck_out_tongue:

Lakshadeep, 2020-06-01:
> I see a few weird thing here:
>
> 1. If this is the SubTree, you are NOT showing me the parent tree where the remapping is done. Remapping is done on the caller side (parent tree), not the callee [...]

Thanks a lot for the quick reply. This resolved the issue.

> That would also be equivalent to this redundant version that set two BB entries (notice the difference with your code):
> 
> ```
>     <DetectPeople name="detect_people" detected_people="{dp}" />
>     <SetBlackboard output_key="detected_people" value="{dp}" />
> ```

Thanks for the example, now I understand the remapping part better!

closed 2020-06-01 by Lakshadeep