BehaviorTree/BehaviorTree.CPP #202: Allow hot reload of trees

facontidavide, 2020-06-01:
Feature: stop the current tree and load a new one from a different file.
Add the corresponding tutorial

SteveMacenski, 2021-02-03:
This is something we'd benefit from on Nav2 - just a note :smile:

simchanu29, 2021-10-05:
I implemented that on my code with a setup function and a ROS service. I'm not sure how this could be integrated into BehaviorTree.CPP (without the ROS part obviously) 

```
    void create_tree(){
        ROS_INFO("Reading tree from %s", path_tree.c_str());
        tree = factory.createTreeFromFile(path_tree);
    }

    void setup(){
        create_tree();

        // This logger publish status changes using ZeroMQ. Used by Groot
        ROS_INFO("[BT] Serving to %d and publishing to %d", zmq_server_port, zmq_publisher_port);
        publisher_zmq.reset();
        publisher_zmq = std::make_unique<BT::PublisherZMQ>(tree, 25, zmq_publisher_port, zmq_server_port);
    }
```

And a service that hot reload the xml file
```
    void tick(){
        if(can_tick){
            tree.tickRoot();
        }
    }

   bool update_behavior_tree(mypkg_msgs::SetString::Request  &req, mypkg_msgs::SetString::Response &res){
        // init
        can_tick = false;
        // ROS_INFO("Behavior tree trying to update with command : %s", req.data.c_str());

        // process
        path_tree = replace_pkg_path(req.data); // custom function to replace $(find mypkg) with correct path
        ROS_INFO("Behavior tree trying to update with file : %s", path_tree.c_str());
        res.message = "Behavior tree updated : "+path_tree;

        try {
            setup();
            res.success = true;
            res.message = "Behavior tree updated : "+path_tree;
        }
        catch(...) {
            ROS_ERROR("[BT] Behavior tree not found : %s", path_tree.c_str());
            res.success = false;
            res.message = "Behavior tree not found : "+path_tree;
        }
        can_tick = true;
        return res.success;
    }
```