BehaviorTree/BehaviorTree.CPP #206: Broken Eloquent Deb Packaging Job

mjcarroll, 2020-06-11:
The Eloquent `deb` build for BehaviorTree has been broken for a bit of time:

http://build.ros2.org/job/Ebin_uB64__behaviortree_cpp_v3__ubuntu_bionic_amd64__binary/

It looks like you have a test_depends on `ament_cmake_gtest`, but then use it without checking for `BUILD_TESTING` here: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/ros2-devel/CMakeLists.txt#L58

```
23:05:38 CMake Error at CMakeLists.txt:74 (find_package):
23:05:38   By not providing "Findament_cmake_gtest.cmake" in CMAKE_MODULE_PATH this
23:05:38   project has asked CMake to find a package configuration file provided by
23:05:38   "ament_cmake_gtest", but CMake did not find one.
23:05:38 
23:05:38   Could not find a package configuration file provided by "ament_cmake_gtest"
23:05:38   with any of the following names:
23:05:38 
23:05:38     ament_cmake_gtestConfig.cmake
23:05:38     ament_cmake_gtest-config.cmake
23:05:38 
23:05:38   Add the installation prefix of "ament_cmake_gtest" to CMAKE_PREFIX_PATH or
23:05:38   set "ament_cmake_gtest_DIR" to a directory containing one of the above
23:05:38   files.  If "ament_cmake_gtest" provides a separate development package or
23:05:38   SDK, be sure it has been installed.
23:05:38 
```

facontidavide, 2020-06-11:
Should be fixed now

closed 2020-06-11 by facontidavide

mjcarroll, 2020-06-11:
Are you going to re-bloom?

facontidavide, 2020-06-11:
https://github.com/ros/rosdistro/pull/25428