BehaviorTree/BehaviorTree.CPP #207: Bug in referencing ROS2 package paths in xml

Serafadam, 2020-06-17:
Hi, I think I've found a bug, in ROS2 version of package, for xml referencing ros1 version of package searching is used, in ROS2 it's `ament_index_cpp::get_package_share_directory` instead of `ros::package::getPath` as per https://answers.ros.org/question/288501/ros2-equivalent-of-rospackagegetpath/

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/47d770c38623be39b677efa0a6e164a9085500e3/src/xml_parsing.cpp#L143

On a side note, is there a plan to release newest version of the code in ROS Dashing?

facontidavide, 2020-06-17:
thanks. I will have a look at it

asasine, 2021-04-23:
I'm not sure this bug would be hit since `USING_ROS` isn't defined when building with ament https://github.com/BehaviorTree/BehaviorTree.CPP/blob/725eba7e0abbc704284dec393706e5404f1472e3/CMakeLists.txt#L74-L76

Using the `ros_pkg` attribute would cause a runtime error in that case since `#ifdef USING_ROS` doesn't pass and `#else` is called https://github.com/BehaviorTree/BehaviorTree.CPP/blob/725eba7e0abbc704284dec393706e5404f1472e3/src/xml_parsing.cpp#L136-L149

closed 2022-03-06 by facontidavide

asasine, 2022-04-08:
> Edit: Also, I have some problems when including files with relative paths, for example to the previous directory `../tree_that_i_want_to_include.xml`

@Serafadam I've encountered the same issue and started #324 to track it. I think I found a resolution and created #358, a review from a maintainer would be good since it does change the existing behavior of tree inclusion.