BehaviorTree/BehaviorTree.CPP #209: Noetic release cannot find coroAction node

yimenghou, 2020-06-28:
Hi,
I try to port my code from ROS Melodic to ROS Noetic.
I used several types of action nodes (Sync, Coro, etc.. ) They all works just fine in Melodic. 
But when I port the same code in Noetic. The compiler complains about missing  'CoroActionNode'.
Then I write a very simple code to test if coroAction work or not in Noetic. 
The code has no compile/link error in Melodic but when switch to Noetic. The compiler outputs the following:
```.
/usr/bin/ld: CMakeFiles/test_foo.dir/src/test_foo.cpp.o: in function `FooActionNode::FooActionNode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, BT::NodeConfiguration const&)':
test_foo.cpp:(.text._ZN13FooActionNodeC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKN2BT17NodeConfigurationE[_ZN13FooActionNodeC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKN2BT17NodeConfigurationE]+0x2b): undefined reference to `BT::CoroActionNode::CoroActionNode(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, BT::NodeConfiguration const&)'
/usr/bin/ld: CMakeFiles/test_foo.dir/src/test_foo.cpp.o: in function `FooActionNode::~FooActionNode()':
test_foo.cpp:(.text._ZN13FooActionNodeD2Ev[_ZN13FooActionNodeD2Ev]+0x14): undefined reference to `BT::CoroActionNode::~CoroActionNode()'
/usr/bin/ld: CMakeFiles/test_foo.dir/src/test_foo.cpp.o:(.rodata._ZTV13FooActionNode[_ZTV13FooActionNode]+0x20): undefined reference to `BT::CoroActionNode::executeTick()'
/usr/bin/ld: CMakeFiles/test_foo.dir/src/test_foo.cpp.o:(.rodata._ZTI13FooActionNode[_ZTI13FooActionNode]+0x10): undefined reference to `typeinfo for BT::CoroActionNode'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [CMakeFiles/test_foo.dir/build.make:92: /home/yimeng/ws_ws/devel/.private/foo_project/lib/foo_project/test_foo] Error 1
make[1]: *** [CMakeFiles/Makefile2:276: CMakeFiles/test_foo.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

Here is my CmakeList.txt file:
```
cmake_minimum_required(VERSION 3.0.2)
project(foo_project)

if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

find_package(catkin REQUIRED COMPONENTS
  behaviortree_cpp_v3
)

catkin_package(
  CATKIN_DEPENDS behaviortree_cpp_v3
)

include_directories(
  ${catkin_INCLUDE_DIRS}
)

add_executable(test_foo
  src/test_foo.cpp
)
target_link_libraries(test_foo ${BEHAVIOR_TREE_LIBRARY}  ${catkin_LIBRARIES})

install(TARGETS test_foo
  RUNTIME DESTINATION ${CATKIN_PACKAGE_BIN_DESTINATION}
)
```

and my test code
```
#include "behaviortree_cpp_v3/bt_factory.h"

// clang-format off
static const char* xml_text = R"(
<root main_tree_to_execute="BehaviorTree">
  <BehaviorTree ID="BehaviorTree">
    <SequenceStar name="root">
      <Action ID="FooActionNode"/>
    </SequenceStar>
  </BehaviorTree>
</root>
 )";

// clang-format on

class FooActionNode : public BT::CoroActionNode {
 public:
  FooActionNode(const std::string& name, const BT::NodeConfiguration& cfg) : BT::CoroActionNode(name, cfg) {}

  virtual BT::NodeStatus tick() override {
    return BT::NodeStatus::SUCCESS;
  };

  virtual void halt() override {};

  // Any subclass of BtActionNode that accepts parameters must provide a
  // providedPorts method
  static BT::PortsList providedPorts() { return {}; }
};

int main(int argc, char** argv) {
  BT::BehaviorTreeFactory factory;
  factory.registerNodeType<FooActionNode>("FooActionNode");
  auto tree = factory.createTreeFromText(xml_text);
  BT::NodeStatus status = BT::NodeStatus::IDLE;
  while(status == BT::NodeStatus::IDLE || status == BT::NodeStatus::RUNNING) {
    status = tree.tickRoot();
    using namespace std::chrono_literals;
    std::this_thread::sleep_for(50ms);
  }
  return 0;
}
```

I get behaviortree_cpp_v3 via 'apt install ros-noetic-behaviortree_cpp_v3', it outputs the following
```
ros-noetic-behaviortree-cpp-v3 is already the newest version (3.5.0-1focal.20200529.055742).
```
The 'uname -a' output:
```
Linux yimeng-lenovo 5.4.0-39-generic #43-Ubuntu SMP Fri Jun 19 10:28:31 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

The 'printenv | grep ROS' output:
```
ROS_DISTRO=noetic
ROS_ETC_DIR=/opt/ros/noetic/etc/ros
ROS_PACKAGE_PATH=/opt/ros/noetic/share
ROS_PYTHON_VERSION=3
ROS_VERSION=1
ROS_ROOT=/opt/ros/noetic/share/ros
ROS_MASTER_URI=http://localhost:11311
ROSLISP_PACKAGE_DIRECTORIES=
```

Thanks ahead for help.

mechatheo, 2020-07-29:
Hello, 

we are also experiencing linker issues on noetic. I checked the shared object file contained in the debian (currently version 3.5.0) and i also built the package myself (newest version 3.5.1). In both cases this happens:

```
nm /opt/ros/noetic/lib/libbehaviortree_cpp_v3.so 
nm: /opt/ros/noetic/lib/libbehaviortree_cpp_v3.so: no symbols
```

on melodic however all symbols are there as expected. So something maybe got messed up regarding symbol visibility

mechatheo, 2020-09-01:
Ok, what i did with `nm` was nonsense, you have to use the `-D` flag to list symbols in a dynamic library in release mode. However i can confirm the linker errors regarding `CoroActionNode` on `noetic` when using the debian package.

facontidavide, 2020-09-02:
I will investigate

Serafadam, 2020-12-10:
I have the same issue on Foxy debs, if I build the repository from source the problem is gone, so maybe a new release could fix this problem

facontidavide, 2021-01-27:
would you please try with version 3.5.5. I am releasing it today

yimenghou, 2021-02-05:
I tried to build my project with this repository switching to tag "3.5.5". It's still not working. Same reason.

js-lft, 2021-02-18:
I seem to have the same issue with ROS Noetic and Ubuntu 20.04.

I also tried in the CMakeLists.txt explicity: `target_link_libraries(test_foo behaviortree_cpp_v3  ${catkin_LIBRARIES})` because that's what is suggested in the 
This gets the package to compile fully.
However, when running the ROS node, it results in a symbol lookup error:
>  undefined symbol: _ZN2BT14CoroActionNode11executeTickEv

With `${BEHAVIOR_TREE_LIBRARY}` or actually without any reference to the BT library in target_link_libraries() I get
> flockfollow.cpp:(.text+0xc62): undefined reference to `BT::CoroActionNode::halt()'
/usr/bin/ld: CMakeFiles/behavior_tree_node.dir/src/nodes/flockfollow.cpp.o:(.data.rel.ro._ZTV11FlockFollow[_ZTV11FlockFollow]+0x20): undefined reference to `BT::CoroActionNode::executeTick()'
/usr/bin/ld: CMakeFiles/behavior_tree_node.dir/src/nodes/flockfollow.cpp.o: in function `FlockFollow::~FlockFollow()':
flockfollow.cpp:(.text._ZN11FlockFollowD2Ev[_ZN11FlockFollowD5Ev]+0x38): undefined reference to `BT::CoroActionNode::~CoroActionNode()'
/usr/bin/ld: CMakeFiles/behavior_tree_node.dir/src/nodes/flockfollow.cpp.o:(.data.rel.ro._ZTI11FlockFollow[_ZTI11FlockFollow]+0x10): undefined reference to `typeinfo for BT::CoroActionNode'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/behavior_tree_node.dir/build.make:459: /home/jsee/dev/ros_ws/devel/.private/jsee_pkg/lib/jsee_pkg/behavior_tree_node] Error 1
make[1]: *** [CMakeFiles/Makefile2:576: CMakeFiles/behavior_tree_node.dir/all] Error 2
make: *** [Makefile:160: all] Error 2

Same results whether by installing the library via apt or by compiling the current GitHub version.

nocoinman, 2021-07-16:
> I seem to have the same issue with ROS Noetic and Ubuntu 20.04.
>
> I also tried in the CMakeLists.txt explicity: `target_link_libraries(test_foo behaviortree_cpp_v3 ${catkin_LIBRARIES})` because [...]

Experiencing the same issue

smitchell7, 2021-09-28:
The issue is that there's some inverted logic built into the action header.
`#ifndef BT_NO_COROUTINES`
This is a double negative, and it should be 
`#ifdef BT_NO_COROUTINES`

okamoto007, 2022-02-15:
You can put the package BehaviorTree.CPP in to catkin_ws/src/, and comment the CMakeLists.txt in BehaviorTree.CPP like this:

#---- Include boost to add coroutines ----
find_package(Boost COMPONENTS coroutine QUIET)
if(Boost_FOUND)
    include_directories(${Boost_INCLUDE_DIRS})
#    if(NOT Boost_VERSION VERSION_LESS 105900)
        message(STATUS "Found boost::coroutine2.")
        add_definitions(-DBT_BOOST_COROUTINE2)
        set(BT_COROUTINES true)
#    elseif(NOT Boost_VERSION VERSION_LESS 105300)
#        message(STATUS "Found boost::coroutine.")
#        include_directories(${Boost_INCLUDE_DIRS})
#        add_definitions(-DBT_BOOST_COROUTINE)
#        set(BT_COROUTINES true)
#    endif()
endif()

and recompile

facontidavide, 2022-11-22:
I found a solution. This should be fixed in version 3.8.1, to be released soon 
562c0e52a7c688b853621d0c798ccc8c8a66e05b

closed 2022-11-22 by facontidavide