BehaviorTree/BehaviorTree.CPP #213: Update release repository for ROS 2 rolling release support

nuclearsandwich, 2020-07-23:
In order to be automatically released from Foxy to Rolling, and in order automatically release from Rolling to future distributions, the rolling release automation needs to be able to push to release repositories.
For this reason, [REP 2002](https://www.ros.org/reps/rep-2002.html#managing-build-farm-writable-release-repositories) specified that release repositories for the Rolling distribution would be created in the [ros2-gbp](https://github.com/ros2-gbp) organization if they were not part of that organization already.

When the Rolling distribution was created, a new release repository for behaviortree_cpp, was created at https://github.com/ros2-gbp/behaviortree_cpp-release

My recommendation would be to switch to using the ros2-gbp release repository for all ROS 2 releases of behaviortree.
With your confirmation, I can add maintainers to the repository in ros2-gbp with full administrative access. Then we would only need to update the release repository entry in other distros where behaviortree_cpp is released.

My recommendation would be to archive the previous release repository as-is, keeping it intact for any sources still using it for earlier releases.

facontidavide, 2020-07-24:
sounds good to me

SteveMacenski, 2023-06-03:
@facontidavide I think you can close this ticket

closed 2023-06-07 by facontidavide