BehaviorTree/BehaviorTree.CPP #218: Library in Debian Package does not have any exported Symbols

mechatheo, 2020-08-13:
Hello,

we wanted to use behaviortree as a debian package instead of building it from source. However the shared library contained in the curent debian packages does not export any symbols, which basically yields it unusable.

```
nm: /opt/ros/noetic/lib/libbehaviortree_cpp_v3.so: no symbols
```

When i build the project from source using `catkin` or plain `CMake` the library does contain all expected symbols. Hoewer when i create a Debian using `bloom-generate` the resulting library is empty. This is so for the package installed from the official Ubuntu repository as for the one i've generated myself using `bloom-generate`.

facontidavide, 2020-08-13:
thank you for reporting. 

I am not really sure how this is happening or why. In which ROS version is this happening?

mechatheo, 2020-08-13:
It hapens on `melodic` and on `noetic` with both ubuntu `Bionic` and `Focal`.

facontidavide, 2020-09-01:
So, I tried to compile a test program as usual on a freshly installed melodic.

BT3 has been installed with the command 

     sudo apt install ros-melodic-behaviortree-cpp-v3

Everything seems to work just fine. Attached there is the test package.

[test_bt.zip](https://github.com/BehaviorTree/BehaviorTree.CPP/files/5154972/test_bt.zip)

mechatheo, 2020-09-01:
ok @facontidavide you right, i tested your example and it worked, i tested it in both versions `melodic` and `noetic` and it worked out. Anyway my whole `nm` thing, was rather stupid anyway, because you need to use `nm -D` to list dynamic symbols anyway. Nevertheless we still have linking issues in our package, it also does use more from the library then your example, so i will continue my investigations.

facontidavide, 2020-09-01:
Once you give me a way to reproduce this, I will be happy to address it.

For the time being I am closing this.

closed 2020-09-01 by facontidavide