BehaviorTree/BehaviorTree.CPP #219: Building agains newer Versions of ZMQ causes deprecation warnings

mechatheo, 2020-08-14:
When building on Focal/Noetic, the following can be observed:

```
Warnings << behaviortree_cpp_v3:make /root/catkin_ws/logs/behaviortree_cpp_v3/build.make.000.log

[2020-08-14T09:09:19.258Z] /root/catkin_ws/src/BehaviorTree.CPP/src/loggers/bt_zmq_publisher.cpp:68:46: warning: 'recv' is deprecated: from 4.3.1, use recv taking a reference to message_t and recv_flags [-Wdeprecated-declarations]

[2020-08-14T09:09:19.258Z]                 bool received = zmq_->server.recv(&req);

[2020-08-14T09:09:19.258Z]                                              ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:1405:5: note: 'recv' has been explicitly marked deprecated here

[2020-08-14T09:09:19.258Z]     ZMQ_CPP11_DEPRECATED(

[2020-08-14T09:09:19.258Z]     ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:67:35: note: expanded from macro 'ZMQ_CPP11_DEPRECATED'

[2020-08-14T09:09:19.258Z] #define ZMQ_CPP11_DEPRECATED(msg) ZMQ_DEPRECATED(msg)

[2020-08-14T09:09:19.258Z]                                   ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:47:31: note: expanded from macro 'ZMQ_DEPRECATED'

[2020-08-14T09:09:19.258Z] #define ZMQ_DEPRECATED(msg) [[deprecated(msg)]]

[2020-08-14T09:09:19.258Z]                               ^

[2020-08-14T09:09:19.258Z] /root/catkin_ws/src/BehaviorTree.CPP/src/loggers/bt_zmq_publisher.cpp:73:34: warning: 'send' is deprecated: from 4.3.1, use send taking message_t and send_flags [-Wdeprecated-declarations]

[2020-08-14T09:09:19.258Z]                     zmq_->server.send(reply);

[2020-08-14T09:09:19.258Z]                                  ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:1325:5: note: 'send' has been explicitly marked deprecated here

[2020-08-14T09:09:19.258Z]     ZMQ_CPP11_DEPRECATED("from 4.3.1, use send taking message_t and send_flags")

[2020-08-14T09:09:19.258Z]     ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:67:35: note: expanded from macro 'ZMQ_CPP11_DEPRECATED'

[2020-08-14T09:09:19.258Z] #define ZMQ_CPP11_DEPRECATED(msg) ZMQ_DEPRECATED(msg)

[2020-08-14T09:09:19.258Z]                                   ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:47:31: note: expanded from macro 'ZMQ_DEPRECATED'

[2020-08-14T09:09:19.258Z] #define ZMQ_DEPRECATED(msg) [[deprecated(msg)]]

[2020-08-14T09:09:19.258Z]                               ^

[2020-08-14T09:09:19.258Z] /root/catkin_ws/src/BehaviorTree.CPP/src/loggers/bt_zmq_publisher.cpp:165:21: warning: 'send' is deprecated: from 4.3.1, use send taking message_t and send_flags [-Wdeprecated-declarations]

[2020-08-14T09:09:19.258Z]     zmq_->publisher.send(message);

[2020-08-14T09:09:19.258Z]                     ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:1325:5: note: 'send' has been explicitly marked deprecated here

[2020-08-14T09:09:19.258Z]     ZMQ_CPP11_DEPRECATED("from 4.3.1, use send taking message_t and send_flags")

[2020-08-14T09:09:19.258Z]     ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:67:35: note: expanded from macro 'ZMQ_CPP11_DEPRECATED'

[2020-08-14T09:09:19.258Z] #define ZMQ_CPP11_DEPRECATED(msg) ZMQ_DEPRECATED(msg)

[2020-08-14T09:09:19.258Z]                                   ^

[2020-08-14T09:09:19.258Z] /usr/include/zmq.hpp:47:31: note: expanded from macro 'ZMQ_DEPRECATED'

[2020-08-14T09:09:19.258Z] #define ZMQ_DEPRECATED(msg) [[deprecated(msg)]]

[2020-08-14T09:09:19.258Z]                               ^

[2020-08-14T09:09:19.258Z] 3 warnings generated.

[2020-08-14T09:09:19.258Z] /root/catkin_ws/src/BehaviorTree.CPP/tools/bt_recorder.cpp:60:24: warning: 'recv' is deprecated: from 4.3.1, use recv taking a reference to message_t and recv_flags [-Wdeprecated-declarations]

[2020-08-14T09:09:19.258Z]             subscriber.recv(&update);

[2020-08-14T09:09:19.258Z]                        ^

[2020-08-14T09:09:19.259Z] /usr/include/zmq.hpp:1405:5: note: 'recv' has been explicitly marked deprecated here

[2020-08-14T09:09:19.259Z]     ZMQ_CPP11_DEPRECATED(

[2020-08-14T09:09:19.259Z]     ^

[2020-08-14T09:09:19.259Z] /usr/include/zmq.hpp:67:35: note: expanded from macro 'ZMQ_CPP11_DEPRECATED'

[2020-08-14T09:09:19.259Z] #define ZMQ_CPP11_DEPRECATED(msg) ZMQ_DEPRECATED(msg)

[2020-08-14T09:09:19.259Z]                                   ^

[2020-08-14T09:09:19.259Z] /usr/include/zmq.hpp:47:31: note: expanded from macro 'ZMQ_DEPRECATED'

[2020-08-14T09:09:19.259Z] #define ZMQ_DEPRECATED(msg) [[deprecated(msg)]]
```

would be nice to have some ifdefs in order to migrade towards the new api of ZMQ.

facontidavide, 2020-09-17:
There is no easy fix, since there is an API change, so either I am compatible with the new one or the old one. A potential solution could be to embed this file in 3rd party.

mechatheo, 2020-09-21:
in our codebase we implemented a custom logger which is compatible on Groot, there we did something like this:

```
     auto success = m_impl->server.recv(
            #if ZMQ_VERSION >= ZMQ_MAKE_VERSION(4, 3, 1)
                            req
            #else
                            &req
            #endif
                            );
                if (success)
                {
                    zmq::message_t reply(tree_buffer.size());
                    memcpy(reply.data(), tree_buffer.data(), tree_buffer.size());
                    m_impl->server.send(reply,
                    #if ZMQ_VERSION >= ZMQ_MAKE_VERSION(4, 3, 1)
                        zmq::send_flags::none
                    #else
                        0
                    #endif
                    );
                }
```

which i understand is not nice and might not be feasible for a larger code base. However one could introduce a wrapper around the send and receive methods and handle the `#ifdef` business there.

facontidavide, 2020-09-21:
looks good to me, I will do the same

facontidavide, 2021-05-02:
Fixed

closed 2021-05-02 by facontidavide