BehaviorTree/BehaviorTree.CPP #223: Sharing more complicated object between multiple nodes

sinaaghli, 2020-09-01:
Apologies for opening an issue but I haven't been able to find an answer online and I'm hoping someone could help here.
I am trying to instantiate an object and then I want multiple asynchronous action nodes to be able to access it properly without facing any race conditions.

I initially thought I would create the object in the main file and add it to the blackboard and later each node would access the object through ports but as I started implementing I realized the object is locked only when getInput/setOutput functions are called and right after that the object would not protected anymore (for example if I was to pass a pointer to the driver around).

To be more precise, my goal was to create a hardware driver(Ethernet) object where multiple nodes could access it and send/receive data. What would be the correct way of implementing this with BT?

SteveMacenski, 2020-09-01:
Not commenting on the BT node part, but you could just wrap the hardware driver with a mutex so that only 1 thing can play with it at a time. Or make it an atomic variable. Though I think the BT is single threaded in all cases so only 1 BT node can be ticked at a time even when one is running in an async node.

ramilmsh, 2020-09-01:
@SteveMacenski isn't that the difference between AsyncNode and CoroActionNode?
One creates a separate thread and the other uses coroutines respectively. (Unless something changed since I last checked in)

From docs:
> An Asynchronous Action has it's own thread.

@sinaaghli 
That being said, I agree with Steve: create a shared mutex/atomic by wrapping it with the object itself, it should work. That is if you need multithreading due to design requirements. The better practice, however, is to use coroutines, unless you have to use threading. In your case, that should fix race conditions as well

facontidavide, 2020-09-01:
Hi,

my suggestion is to avoid (ab)using the blackboard.

Even if blackboard can contain any object, there is a tutorial that explain how to pass an object or argument to your nodes without the blackboard.

https://www.behaviortree.dev/tutorial_08_additional_args/

I agree with @SteveMacenski  and @ramilmsh about the use of a mutex.

sinaaghli, 2020-09-01:
Thank you very much @SteveMacenski @ramilmsh @facontidavide these are very helpful.
I wasn't sure if this feature was considered in BT and somehow I missed the tutorial 8 in the webpage. 
Now I know what to do. Thanks again 🙏

closed 2020-09-01 by sinaaghli