BehaviorTree/BehaviorTree.CPP #226: [docs]: Remapping ports between Trees and SubTrees

ValerioMa, 2020-09-02:
In the documentation about the __Remapping ports between Trees and SubTrees__ ([link](https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/docs/tutorial_06_subtree_ports.md)) the remapping methodology explained in the bullet list referring to _words internal/external_ does not match the proposed xml.
Maybe the one in proposed in the xml is the compact representation? In this case, where can I find the explicit version?

facontidavide, 2020-09-02:
I think some image can better explain this concept, that is a little hard to grasp. I should consider reorganizing this tutorial

Aposhian, 2021-11-23:
This is definitely a pain point when mixing ports with direct blackboard access. Updated documentation would help a lot.

facontidavide, 2022-11-22:
Fixed in new documentation. otherwise, let me know

closed 2022-11-22 by facontidavide