BehaviorTree/BehaviorTree.CPP #228: StatefulActionNode behavior inconsequent?

scgroot, 2020-09-10:
Currently, the StatefulActionNode will remain in state SUCCESS/FAILURE until some exteral callers sets it back to IDLE. This means that when it's used within a RepeatNode for instance, it will only runs one, as the RepeatNode does not reset the state.

On the other hand, the SequenceNode does reset the state of it's children after it completed it's sequence. This leads to in-consequent behavior imho.

What is expected behavior regarding nodes that completed (SUCCES/FAILURE)? Is it the responsibility of the parent to reset the child, or should the StatefulAccessNode return to idle on it's own?

facontidavide, 2020-09-10:
The error is in RepeatNode.

Asynchronous Actions should not set themselves to IDLE. Their parent should; the reason is not apparent, but if you don't follow this rule, Reactive Sequence will fail miserably.

Let me see if I can fix RepeatNode

scgroot, 2020-09-10:
'Fail miserably' can be argued about i guess? 

ReactiveSequence says, 'tick these every time', but now the StatefulActionNode will basically swallow the tick, resulting in basically the same behavior as a regular Sequence.

I actually expected the onStart the called again when using a StatefulActionNode in a ReactiveSequence. 
In my example the StatefulActionNode would check some state in onStart, if it's OK it can return SUCCES, if not it returns RUNNING and corrects the state. By using a ReactiveSequence, I basically say, continue checking this state, while doing the rest of the sequence. Now this behaves like a normal Sequence though.

facontidavide, 2020-09-10:
It is... complicated.

Actually I should add a unit test to prove my point.

ONLY asynchronous actions need to be set to IDLE to be tick again. Therefore, Conditions and SyncActions **are** tick at each loop by a ReactiveSequence.

Suppose that a ReactiveSequence has 3 children (A, B, C), all of them async Actions (this includes StatefulActionNode) and that the "do not restart unless in IDLE" is disabled.

You will notice that at some point A-B will be both running simultaneously, instead of sequentially and eventually, A, B, and C will all be RUNNING.

What I suggest is that you create a unit test, play with it and tell me if I am wrong or if there is a better pattern, but I can tell you that this was something I carefully tried to address.

scgroot, 2020-09-10:
hmm, maybe I'm looking at the wrong thing...

StatefulActionNode is not an AsyncAction righ? It just extends from ActionNodeBase.

I guess my question is, what is expected when I tick a StatefulActionNode that previously returned SUCCESS or FAILURE? Most other nodes will be executed again. Now, StatefulActionNode will just return it's previous state and will not be ticked.

Some nodes (eg SequenceNode) reset the state, which 'hides' this behavior.

What if you would simply implement StatefulActionNode::tick it as following (bit simplified):
```cpp
if (status() == RUNNING) {
   onRunning();
} else {
  onStart();
}
```
That of course does change the behavior, mostly when used in a ReactiveSequence, but it would act more as regular nodes?

facontidavide, 2020-09-10:
Any action that may return RUNNING is asynchronous. AsyncAction is only a specific implementation, but CoroAction and StatefulAction are also asynchronous.

The bug you just reported affect all of them, not just StatefulActionNode. I was able to reproduce it with AsyncAction too.

**Anyway, I just added a unit test and I pushed the fix, so your issue is solved.**

closed 2020-09-10 by facontidavide

facontidavide, 2020-09-10:
FYI: https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/tests/gtest_decorator.cpp#L135-L164