BehaviorTree/BehaviorTree.CPP #230: CONSIDER BACKWARD COMPATIBILITY!!!

aufheben1, 2020-09-16:
Many updates of this library does not consider backward compatibility and making issues.

Recently, I've updated ros2 version from eloquent to foxy and BehaviorTree.CPP version has been updated from 3.5.1 to 3.5.3.
This update includes the syntax change in Parallel control node. (#216 )
As a result, when I try to load xml file which uses parallel node, it makes segmentation fault.

I can modify xml myself, but the problem is that all my coworker should update their ros2 version to foxy, which requires Ubuntu upgrade.

In the further update, please consider backward compatibility of this library.

facontidavide, 2020-09-16:
thanks for reporting the issue.

It was not supposed to segfault. I need to check if it a bug .

facontidavide, 2020-09-16:
done

closed 2020-09-16 by facontidavide