BehaviorTree/BehaviorTree.CPP #232: The iterator to fail

Zeno-X, 2020-09-29:
The class Signal member function notify(CallableArgs... args) call std::vector.erase(iterator) then will cause the iterator to fail.

    void notify(CallableArgs... args)
    {
        for (size_t i = 0; i < subscribers_.size();)
        {
            if (auto sub = subscribers_[i].lock())
            {
                (*sub)(args...);
                i++;
            }
            else
            {
                subscribers_.erase(subscribers_.begin() + i);
            }
        }
    }

facontidavide, 2020-09-29:
Have you checked that this is the case? I  tested the code and it works just fine

facontidavide, 2020-09-29:
Works exactly as expected
``` C++
int main()
{
    std::vector<int> v;
    for (int i=0; i<10; i++) {
        v.push_back(i);
    }

    for (size_t i = 0; i < v.size(); ){
        if (v[i] % 2 == 0)
        {
            i++;
        }
        else {
            v.erase(v.begin() + i);
        }
    }

    for (const int  val: v) {
        std::cout << val << std::endl;
    }
    return 0;
}

```

closed 2020-09-29 by facontidavide