BehaviorTree/BehaviorTree.CPP #233: Tick rate pattern

ramilmsh, 2020-09-30:
@facontidavide This is probably a stupid question and I am thinking about it wrong, but I have a concern about tick rate. Suppose I have a set of asynchronous nodes, which take a while to complete. So, on each tick, I simply check completion and yield execution. However, this approach poses a problem that there are no sleeps to avoid excessive ticks (if a node has not been completed at T it will not be completed at T + a couple of nanoseconds) and prevent the program from clogging an entire core.

I am struggling to find a good answer to that. I have simply introduced sleeps into the tick of all asynchronous nodes, however, if my tree was wide this would reduce tick rate dramatically. Could you tell me what the intended implementation would be in this case?

facontidavide, 2020-10-08:
You should not sleep in the node, but rather before /after calling tickRoot()

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/examples/t05_crossdoor.cpp#L95

facontidavide, 2020-10-08:
As an alternative to a fixed Sleep like the one I suggested, **if** you have a backend in another thread that knows that one or more nodes received their  reply, you may use instead 
[std::condition_variable](https://en.cppreference.com/w/cpp/thread/condition_variable).

In particular, instead of the fixed sleep of 1 ms, use **std::condition_variable::wait_for** and send from the backend a signal with **std::condition_variable::notify_all**.

But this is advanced users only ;)

closed 2020-10-08 by facontidavide