BehaviorTree/BehaviorTree.CPP #235: Add the __shared_blackboard option to SubTreePlus

Acwok, 2020-10-08:
Hello @facontidavide,

Would it be possible to add the `__shared_blackboard` option to the `SubTreePlus` ?
We need to have both port remapping and access to global variables.

It would greatly simplify the tree as we won't have to repeat all the setBB.

See the following example:
```xml
<root main_tree_to_execute="BehaviorTree">
<!--  //////////  -->
<BehaviorTree ID="BehaviorTree">
<Sequence>
<SetBlackboard output_key="approach_angle" value="60"/>
<SetBlackboard output_key="approach_offset" value="50"/>
<SetBlackboard output_key="goal_si" value="TILE_1_SI_1"/>
<SetBlackboard output_key="max_velocity_scaling" value="0.8"/>
<SubTree ID="AssembleInnerRingUp" __shared_blackboard="true"/>
</Sequence>
</BehaviorTree>
<!--  //////////  -->
<BehaviorTree ID="AssembleInnerRingUp">
<Sequence>
<SetBlackboard output_key="tile_si" value="TILE_0_SI_1"/>
<SubTree ID="GrabTileFromContainer" clear_arm_from_si="TILE_0_SI_4" ee_si="ARM_SI_0" list_si_pair="ARM_SI_0;TILE_0_SI_1" tile_si="{tile_si}" __shared_blackboard="true"/>
</Sequence>
</BehaviorTree>
<BehaviorTree ID="GrabTileFromContainer">
<Sequence>
<Action ID="MoveArmToSI" approach_angle="0" approach_offset="{approach_offset}" ee_si="{ee_si}" goal_si="{tile_si}"/>
</Sequence>
</BehaviorTree>
<!--  //////////  -->
<TreeNodesModel>
<SubTree ID="AssembleInnerRingUp">
<input_port default="{approach_offset}" name="approach_offset"/>
<input_port default="{clear_offset}" name="clear_offset"/>
</SubTree>
<SubTree ID="GrabTileFromContainer">
<input_port name="clear_arm_from_si"/>
<input_port default="ARM_SI_0" name="ee_si"/>
<input_port name="list_si_pair"/>
<input_port name="tile_si"/>
</SubTree>
<Action ID="MoveArmToSI">
<input_port default="0" name="approach_angle"/>
<input_port default="{approach_offset}" name="approach_offset"/>
<input_port default="{ee_si}" name="ee_si"/>
<input_port default="{max_velocity_scaling}" name="max_velocity_scaling"/>
<input_port name="goal_si"/>
</Action>
</TreeNodesModel>
<!--  //////////  -->
</root>
```

In this example, we correctly fetch the global variables but we cannot remap the `ee_si` port from `MoveArmToSI` to its parent as we are using SubTree instead of SubTreePlus.

facontidavide, 2020-10-08:
Check the comments. The parameter that is equivalent to that is 

     __autoremap="1"

https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/include/behaviortree_cpp_v3/decorators/subtree_node.h#L75-L77

Acwok, 2020-10-09:
Ok we are going to test if we can put additional parameters besides __autoremap in SubTreePlus.

ashwinsushil, 2021-02-04:
@Acwok Did the setting `__autoremap="1"` in SubTreePlus do the trick of accessing the blackboard variable?

I'm using `v3.5.3` and the blackboard variables are not shared (in the subtree) when using SubTreePlus.

facontidavide, 2021-02-26:
shall I close this?

Acwok, 2021-03-11:
Yes, I won't have time to test.
I'm passing the parameters as inputs to the subtrees instead.

closed 2021-03-11 by Acwok