BehaviorTree/BehaviorTree.CPP #236: Reactive Sequences with AsyncActionNodes

lstelzner, 2020-10-12:
I have approximately the following:
``` 
<ReactiveSequence>
  <IsSafe/>
  <Fallback>
     ...additional logic that may interrupt a run
  </Fallback>
  <Sequence name="run">
     <GotRunSignal/>
     ...
    <SimpleActionClientNode/>
  </Sequence>
</ReactiveSequence>
```

SimpleActionClientNode is an AsyncActionNode , its tick() function sends an ActionGoal (ROS) and loops, waiting for a result from the ActionServer or for an `aborted` boolean to be set true in the halt() function. It returns FAILURE if it gets halted/aborted. When halt() gets called I've confirmed that SimpleActionClientNode::tick() returns FAILURE, but the next time the sequence should run SimpleActionClientNode ::tick() the code never gets called.   In the debugger I saw that  in the executeTick() function below that
```
NodeStatus BT::AsyncActionNode::executeTick()
{
    //send signal to other thread.
    // The other thread is in charge for changing the status
    if (status() == NodeStatus::IDLE)
    {
```
status() is returning FAILURE from the last run instead of having gotten reset to IDLE.   The second time the sequence "run" is getting ticked, then SimpleActionClientNode::tick() will get called (I'm assuming something reset its status the previous time it ran through  Sequence "run").  Is there a way to make sure it gets reset to IDLE after its halted FAILURE? Is it incorrect to pair an AsyncActionNode within a ReactiveSequence this way?

lstelzner, 2020-10-12:
I need to confirm with some more testing, but it looks like adding ` BT::AsyncActionNode::halt();`  to my SimpleActionClientNode::halt() function fixed it.  I found this mentioned after reading the comments in `action_node.h`.  You might want to update https://www.behaviortree.dev/tutorial_04_sequence_star/ tutorial to reflect this key aspect.

SegfaultCreator, 2021-07-20:
I came across the same "Problem" when i wondered why the Node Status remained false in the Groot Monitoring . After some research in the documentation i found this description in https://www.behaviortree.dev/tutorial_03_generic_ports/

> You are responsible for the implementation of a valid halt() functionality.

I found it pretty obvious, that i had to add BT::AsyncActionNode::halt() below my own lines of code in the _halt_ function, but maybe an example of such a _halt_ definition could be useful in the documentation.

facontidavide, 2023-10-16:
Assuming his error is obsolete/stale. Closing

closed 2023-10-16 by facontidavide