BehaviorTree/BehaviorTree.CPP #237: Brainstorming: Is the derived class "LeafNode" necessary?

MaxPBauer, 2020-10-20:
Out of curiosity: I noticed that the ActionNode and the ConditionNode inherits from the LeafNode, but this information is not used further.

Under the assumption that _"Multiple inheritance is especially problematic, because it often imposes a higher performance overhead"_, could one omit the LeafNode?
https://google.github.io/styleguide/cppguide.html#Inheritance

This could reduce inheritance depth.

facontidavide, 2020-10-20:
ConditionNode is also a LeafNode

closed 2020-10-20 by facontidavide

MaxPBauer, 2020-10-22:
Sorry, your are absolutely right! I rushed/misread the text, as I thought there was once a statement to reduce multiple inheritance **levels** (Seems like there is/ was no such statement ?!?). 

Still I am wondering if there is a benefit of this extra layer. From what I found with my limited insight, It seems to be unused.

Background: I have quite a different use case and need a minimal behavior tree, but try to keep it as much as possible to an OpenSource implementation (this one I guess). Thus I removed in my fork all dispensable parts and I stumbled over this one.

facontidavide, 2020-10-22:
Unless I am proving wrong, in this particular case the extra layer in the polymorphism doesn't have any cost neither in terms of memory / code bloat / cpu usage.

You have a virtual table and the number of indirection should not change. Therefore this whole discussion is kind of pointless, in my opinion. But of course I will be happy to stand corrected if a benchmark will show me that I am wrong :smile:

facontidavide, 2020-10-22:
So, the question is not just "is there a benefit in this extra layer" but "is there a COST in this extra layer"?

MaxPBauer, 2020-10-22:
From my perspective the "cost" of very deep inheritance  is reduced readability for new users rather than performance issues. But I think we can agree that this opinion is subjective/ less relevant here.

As we are already talking about implementation details, I hope you dont mind if I bring up some other small questions, otherwise feel free to ignore :) I do understand this as a kind of brainstorming rather a concrete issue, as it is not urgent and it would implicate some work.  

**1. Usage of raw pointers for children nodes**
Is it intended to keep the raw pointers for children behaviors? I struggle with the idea of letting the users handle raw pointers.. Wouldn't the TreeNode::Ptr be the better solution?

**Separation of concerns between tree nodes and blackboard**
Even I really love the blackboard implementation, I maybe need to remove it in my fork (at least for some time).
I noticed here that the TreeNode has a few methods such as TreeNode::getInput(..) and more, which could be separated into the configs_. This way it would be easily possible to have "naked" behavior trees and easily composed behavior trees with blackboards. Whats your opinion on that?

facontidavide, 2020-10-22:
No problem, I love discussing different opinions :smile: 

1) Raw pointers. These are non owning pointers. I COULD use **std:.weap_ptr** to keep a similar semantic, but child are never deleted, because I oppose strongly to any dynamic change in the tree. I will never facilitate this dangerous anti-pattern.
The user is NOT responsible to manage the children, as you implied.

2) Separation of concerns: there is a very compelling reason to use the abstract concept of "ports" instead of blackboard and that is the scope of the blackboard, to avoid collisions in the blackboard. If you haven't worked long enough with this, you may not understand or underestimate the problem.

In short: Tree and Blackboard are tightly coupled because:

- People hide the blackboard from the model, while BT.CPP does a gigantic effort to make that model visible for very good reasons.

- The blackboard model used by other libraries is potentially wrong, because usually BB of simplistic implementations don't have. I provide an API that force the user to use BB correctly and safely.

So, you are of course free to "simplify" your behavior tree, but keep in mind that this library is the results of 3 iterations over 1 year of work, where the first iteration was "simpler", but not scalable.

I suggest you to read this document:  https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/MOOD2Be_final_report.pdf

MaxPBauer, 2020-10-22:
Ok, I understand the design decisions now, especially for your intended use case.
Thanks for the clarification!