BehaviorTree/BehaviorTree.CPP #240: Reading entries from the blackboard right after a tick fails occasionally

matthias-mayr, 2020-10-29:
I am using this library in reinforcement learning to represent my policy. For every set of parameters of an episode, I create a new tree, add the parameters to the blackboard and use them in some of the nodes. Some nodes also write entries to the blackboard that I am reading after the tick.

I observed that sometimes, but especially after the first tick, I am not able to retrieve that parameters that should have been set. Although my tree should always write them. Here's a stripped-down version of my tree:
```xml
 <root main_tree_to_execute = "ObstacleTree" >
     <BehaviorTree ID="ObstacleTree">
        <ReactiveFallback name="root_sequence">
            <ReactiveSequence name="3-goal">
                <EEYThreshold   name="ee_y_threshold" threshold="{y_threshold}" ee_y="{ee_y}"/> 
                <SetMGGoal   name="mg_goal3"
                    mp_param_x="{mp_3_x}"
                    mg_trans_x="{mg_trans_x}"
            </ReactiveSequence>
            <ReactiveSequence name="2-traverse">
                <EEZThreshold   name="ee_z_threshold" threshold="{z_threshold}" ee_z="{ee_z}"/> 
                <SetMGGoal   name="mg_goal2"
                    mp_param_x="{mp_2_x}"
                    mg_trans_x="{mg_trans_x}"
            </ReactiveSequence>
            <SetMGGoal   name="mg_goal1"
                    mp_param_x="{mp_1_x}"
                    mg_trans_x="{mg_trans_x}"
        </ReactiveFallback>
     </BehaviorTree>
 </root>
```
By using a reactive fallback, one of the `SetMGGoal` nodes should always be ticked. Their implementation looks like this:
```cpp
class SetMGGoal : public BT::AsyncActionNode
{
public:
    SetMGGoal(const std::string& name, const BT::NodeConfiguration& config) :
        BT::AsyncActionNode(name, config){}

    static BT::PortsList providedPorts() { return {
        BT::InputPort<double>("mp_param_x"),
        BT::OutputPort<double>("mg_trans_x"),
    }; }

    BT::NodeStatus tick() override
    {
        BT::Optional<double> param_x = getInput<double>("mp_param_x");
        if (!param_x)
        {
            throw BT::RuntimeError("missing required input [mg_param_x] ");
        }
        setOutput("mg_trans_x", param_x.value());
        return BT::NodeStatus::RUNNING;
    }
};
```
The essential parts of the creation look like this:
```cpp
BT::BehaviorTreeFactory factory;
factory.registerNodeType<SetMGGoal>("SetMGGoal");
factory.registerNodeType<EEYThreshold>("EEYThreshold");
factory.registerNodeType<EEZThreshold>("EEZThreshold");

_tree = std::make_shared<BT::Tree>(factory.createTreeFromFile("./res/bt_obstacle.xml"));
_tree->rootBlackboard()->set("z_threshold", _params[0]);
_tree->rootBlackboard()->set("y_threshold", _params[1]);
_tree->rootBlackboard()->set("mp_1_x", params[2]);
_tree->rootBlackboard()->set("mp_2_x", params[3]);
_tree->rootBlackboard()->set("mp_3_x", params[4]);
```
And ticking looks like this:
```cpp
_tree->rootBlackboard()->set("ee_y", ee_pose.position.y);
_tree->rootBlackboard()->set("ee_z", ee_pose.position.z);

BT::NodeStatus status = _tree->tickRoot();

Eigen::VectorXd action{Eigen::VectorXd::Zero(7)};
try {
    _tree->rootBlackboard()->get("mg_trans_x", action[0]);
} catch (const std::runtime_error& e) {
    LOG(ERROR) << "Could not retrieve MG config from blackboard.";
}
```
My understanding would be that I should never get an exception. If I observe one after the first tick, I can read out the correct value on the second tick. Any idea why this could be the case?

facontidavide, 2020-10-30:
Let me see....

**mg_trans_x** is set by the **SetMGGoal**. 

Could be that EEYThreshold returns RUNNING? I guess not.

Can you send me an source code that I can compile to debug this myself?

matthias-mayr, 2020-10-30:
> Could be that EEYThreshold returns RUNNING? I guess not.

No, it's a `ConditionNode` and returns only `SUCCESS` or `FAILURE`.

> Can you send me an source code that I can compile to debug this myself?

Sure, I can take it out and make it compile separately. I hope it's still reproducible then, we'll see. Give me a minute.

matthias-mayr, 2020-10-30:
Here is a minimal example: https://github.com/matthias-mayr/bt_minimal_example
It became a catkin package, because that was easiest to build. Also, I am using the ROS binary package in `melodic`, so it's closest to what I am actually running.

It seems like there is a general delay. I can never read the values directly after the first tick. Sometimes also not after the second tick.
When I change the content of `mg_trans_x` as in tick 4 and 5 in the example, this also takes 1 or 2 ticks to be read from the blackboard.

Let me know if I can do something else to shine some light on this.

facontidavide, 2020-11-03:
Mystery solved.

First of all, double check your code:

```
if (ee_y.value() > threshold.value()) {
    return BT::NodeStatus::SUCCESS;
} else {
    return BT::NodeStatus::FAILURE;
}
```

The condition is false (ee_y is 0.0 and threshold is 2.0). Therefore the action **mg_goal3** is never executed as IT SHOULD BE!

Byt there is potentially another subtle problem.

**You are using AsycActionNode is the wrong way**.  AsycActionNode will launch a new thread and execute your tick() asynchronously in that thread. Therefore you will most probably check the blackboard **before** the variable `mg_trans_x` is written.

Furthermore, you can not distinguish between the case when:

 - SetMGGoal::tick() has not been executed yet, because it is running in a different thread.
 - SetMGGoal::tick() is completed and returns RUNNING ? (why, that looks wrong!)

The first step would be to use **BT::ActionNodeBase** instead of **BT::AsycActionNode**.

facontidavide, 2020-11-03:
I am closing the issue, because clearly this is not a bug, but a wrong implementation

closed 2020-11-03 by facontidavide

matthias-mayr, 2020-11-03:
Thanks for looking into this.
> First of all, double check your code:
> The condition is false (ee_y is 0.0 and threshold is 2.0). Therefore the action mg_goal3 is never executed as IT SHOULD BE!

This is true for the first three ticks in the example. This is why I am expecting the `mg_goal1` leaf to set `3`. 
For the last two ticks, I am changing [the content of state[0] to 5](https://github.com/matthias-mayr/bt_minimal_example/blob/master/src/bt_minimal_example.cpp#L21) and then this condition reads `ee_y 5.0 > threshold 2.0` and is true.

> You are using AsycActionNode is the wrong way. AsycActionNode will launch a new thread and execute your tick() asynchronously in that thread. Therefore you will most probably check the blackboard before the variable mg_trans_x is written.
> The first step would be to use BT::ActionNodeBase instead of BT::AsycActionNode.

Thanks a lot for pointing out `ActionNodeBase`. I did not know that this is an option. When looking at the `action_node.h` header file and among the derived classes that allow me to return RUNNING, `AsyncActionNode` seemed to be the closest to what I want to do.

I changed the inheritance for `SetMGGoal` to `ActionNodeBase` in [this subsequent commit](https://github.com/matthias-mayr/bt_minimal_example/commit/abe5cc1f4e83642a8920dd365ffbad25f93d5998) and it resolves the issue. Thanks a lot!

> Furthermore, you can not distinguish between the case when:
>    SetMGGoal::tick() has not been executed yet, because it is running in a different thread.

One thing I had in mind was to pass some iterator variable, that would increase every time I tick and set it in addition to the value of `mg_trans_x`. Then I could check if the incremented iterator has been set on the blackboard already and would know if `mg_trans_x` should be updated. But this workaround is not needed anymore, since using `ActionNodeBase` seems to be the correct solution to this.

> SetMGGoal::tick() is completed and returns RUNNING (why? that looks wrong!)

The parts of the BT that are included in the example only change the motion configuration of the robot. So `SetMGGoal` basically reports that this motion is running. I fetch this motion configuration from the blackboard and execute on the simulated or real robot.
In the real BT, there is a separate condition node that checks if the goal of the BT has been reached and then the whole tree would succeed. I did not include this in the minimal example, so the BT I included would indeed always return RUNNING and nothing else.

Last but not least, thanks a lot for this library and the support. I really appreciate it.

facontidavide, 2020-11-03:
Anyway, just as a reminder. The only use case where **AsyncActionNode** makes sense is when you call a blocking function that may take a long time to be executed and you don't want to block.

It is very easy to do it right, therefore I think I will deprecate it in the future.

matthias-mayr, 2020-11-03:
I assume that you wanted to write 
```
It is very HARD to do it right, therefore I think I will deprecate it in the future.
```
Maybe an alternative would be to write a page in the documentation about the usage of the different types of nodes. It  would of course be a repetition of the comments in the header file, but it also allows to write a bit more and would probably feel more accessible to users than going to the header file.
I did not really need `AsyncActionNode` and just picked it, because I did not know about that `ActionNodeBase` is a viable option. 
Going back to the header file, `ActionNodeBase` says:
```
@brief The ActionNodeBase is the base class to use to create any kind of action.
 A particular derived class is free to override executeTick() as needed.
```
I read this as "this class exists to get inheritance right" with a slight notion of "if you want to create your own **type** of action node, start here". Without your hint, I would not have used it as a base type for my own nodes.

Regarding the `AsyncActionNode` specifically: Maybe some users would find the tutorial about it more easily if it's name in the TOC `Tutorial 4: Sequence` also mentions `AsyncActionNode` like the heading does.

facontidavide, 2020-11-03:
ok, I will consider updating the documentation, thanks