BehaviorTree/BehaviorTree.CPP #241: Behavior of SequenceStar Node

SegfaultCreator, 2020-11-04:
Hello again,

I'm currently struggling to create Behavior with the Sequence Star Node. I want to use the Sequence Star in a way similar to the example in the documentation (see Below).

[image]

What i want to do are the Tasks "FlyToPoint" and "Act at Point" , and when certain conditions are met, i want the active task to be interrupted by the Task "Do Something between" and then go back to the previous Behaviour.
However, the Finished status of The FlyToPoint Node gets reset any time i'm leaving the Sequence Star to perform the "DoSthBetween" task. All Tasks used are ASyncActionNodes.

[image]

Can the Sequence Star only save its internal State if the node left to it has no more children, is it a bug or did i get something wrong with the Documentation?

Thanks in Advance
SegfaultCreator

facontidavide, 2020-11-04:
The problem encountered by you and other people is that there is no concept of "suspending" in Sequence and Fallback (either in Reactive and Star versions).

Let me think what a correct aptterns would be in this case.

SegfaultCreator, 2020-11-04:
Ah, so you mean the only difference between Sequences and Sequence star is the Failure case, which triggers "Tick Again"? 
I mean your Table at the top displays it well but i got a bit misinterpreted the discription saying that the first node is not ticked again. 
I think it could be quite useful to have these "permanent states", for example to model behavior of game ai. (e.g. Bot that does its procedure unless it is disturbed)

closed 2023-06-07 by facontidavide