BehaviorTree/BehaviorTree.CPP #242: Reactive Sequence Concept

isuru-madara, 2020-11-20:
Hi, This isn't strictly an issue although it has been bugging me for few days so I thought I'd ask it here. 

First of all, this is an amazing library to add behaviour trees to ROS, so thank you... 
I have been developing a simple patrolling behaviour for my robot. I have taken the example given in tutorials under "Learn the basics->Sequence Node" ( https://www.behaviortree.dev/sequencenode/ ) and extended it with a Fallback node so that the final tree looks something as below. 

[image]

(Note: All GoTo nodes are MoveBaseActions inherited  from on AsyncActionNode)

It works just as I hope it would. (Robot moves to destinations A, B and C in order. If a battery low message arrives during navigation, it goes to the charging dock) However, In the class reference documentation it says Reactive sequence should only have a single Asynchronous child to work properly. (http://docs.ros.org/en/kinetic/api/behaviortree_cpp_v3/html/classBT_1_1ReactiveSequence.html)

So my question is, wouldn't the children of Sequence Star node count as children of Reactive sequence ? is it because they're not directly a child of the Reactive sequence node ?

ramilmsh, 2020-11-21:
@isuru-madara yes, only direct children count as children.

This is done for a good reason: modularity. Behavior tree is an abstraction, where you can define complex behavior, by using simple building blocks (action nodes) and combining them with the control nodes, but that's only how we perceive it. Now, for the tree itself there is no difference between those - there is only an interface: Node, i does not matter what the node does or how, it only matters that it has ::tick(), which will return a status (SUCCESS, FAILURE or RUNNING). And, in this case, async action will be equivalent to sequence star, because both comply with the same interface, so ticking one or the other is processed by the same logic, which is not even aware of different types of nodes, simply that they are nodes.

A little convoluted, but hopefully makes sense.

isuru-madara, 2020-11-23:
@ramilmsh It actually does make sense now. Thank you very much.

closed 2020-11-23 by isuru-madara