BehaviorTree/BehaviorTree.CPP #243: Dose port support STL::Container?

jiaobendaye, 2020-11-24:
HI,
  I am using BehaviorTree.CPP to refactor my old AI Bot. But, there is some complex data, such as std::vector<std::vector<...>>, which I have no idea how to implement convertFromString for using port. Maybe I could use yaml-cpp to implement it. Could you give me  some suggestions? Thanks a lot!

facontidavide, 2020-11-24:
Hi,

Ports support **any** C++ object, including std containeres.

Please remember that implementing `convertFromString`is **optional** and required **only** when you want to initialize its value from a string in the XML.

The way you implement it is completely irrelevant for BT.CPP, since **you** will (if needed) implement it.

closed 2020-11-24 by facontidavide

jiaobendaye, 2020-11-25:
get it! Tks again!