BehaviorTree/BehaviorTree.CPP #251: Tutorial 01 - unclear instructions

samurairepo, 2021-01-14:
Hello, I failed to reproduce the first tutorial, mainly because I think it is missing some steps, which might be obvious to more experienced developer.

Attempted tutorial:
https://www.behaviortree.dev/tutorial_01_first_tree/

Questions:
This class declaration, should be in "dummy_nodes.h" correct ?

```
// Example of custom SyncActionNode (synchronous action)
// without ports.
class ApproachObject : public BT::SyncActionNode
{
  public:
    ApproachObject(const std::string& name) :
        BT::SyncActionNode(name, {})
    {
    }

    // You must override the virtual function tick()
    BT::NodeStatus tick() override
    {
        std::cout << "ApproachObject: " << this->name() << std::endl;
        return BT::NodeStatus::SUCCESS;
    }
};
```
Main:
The header does not include all the classes that are in the next source code, but I understand it is my job to fill their declaration. I removed the undeclared ones instead.
```
#include "behaviortree_cpp_v3/bt_factory.h"

// file that contains the custom nodes definitions
#include "dummy_nodes.h"

int main()
{
    // We use the BehaviorTreeFactory to register our custom nodes
    BehaviorTreeFactory factory;

    // Note: the name used to register should be the same used in the XML.
    using namespace DummyNodes;

    // The recommended way to create a Node is through inheritance.
    factory.registerNodeType<ApproachObject>("ApproachObject");

    // TreeNodes are destroyed
    auto tree = factory.createTreeFromFile("./my_tree.xml");

    tree.tickRoot();

    return 0;
}
Please, notice that in the cpp file there is "using namespace DummyNodes;", but in the header from the same tutorial, there is no namespace DummyNodes. Should I add DummyNodes in the header

```
Now, from the Github page, I  found this CMAKE file (it is not in the Tutorial 01).
What should be the correct CMAKE for this tutorial ?
How do I include XML file in CMAKE build ?

```
cmake_minimum_required(VERSION 3.5)

project(hello_BT)

set(CMAKE_CXX_STANDARD 14)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
find_package(BehaviorTreeV3)

add_executable(${PROJECT_NAME} "hello_BT.cpp")
target_link_libraries(${PROJECT_NAME} BT::behaviortree_cpp_v3)
```
Lastly, I searched the Github folder with examples and another thing confuses me, which is, that the tutorial source codes are very different.
In Github main, for example, the tree is represented as string literal, but in the landing page, the tree is represented like extrernal XML file. 
Also this line: "factory.registerFromPlugin("./libdummy_nodes_dyn.so");"
It is not in the landing page tutorial, but it is in github.. what is "libdummy_nodes_dyn.so" ? Where can I get this file ? What is its purpose ?

Please can you help me and future beginners. **Please, can you provide sequential steps, with fully functional example ?** I admit, it is probably due to my own lack of experience, but I am very lost, I have spent hours trying to make tutorial 01 work, with no success. Thank you, Tomas

ramilmsh, 2021-01-15:
Ok, so, first: ignore "factory.registerFromPlugin("./libdummy_nodes_dyn.so");", it will not be helpful for now.

Second, the reason the tree is loaded sometimes from an xml file and sometimes from a string, is that the xml file contains the same data as the string, except, it's in a file. This is a matter of convenience, for tutorials - it's simpler to keep it in a string, but once you start using Groot or doing production - xml is the way to go.

Thrid, do not change dummy_nodes.h, it works just fine

Could you upload your project to github and share the link, so I may look at it and find any issues?

TomasHricina, 2021-01-15:
First of all, thank you for your reply. 

I am struggling to compose minimal working example.
Here is a try: https://github.com/TomasHricina/BehaviorTrees/tree/main

Thank you, Tomas

facontidavide, 2021-01-15:
I will look at it tomorrow, if you can wait. Too busy today

TomasHricina, 2021-01-15:
> I will look at it tomorrow, if you can wait. Too busy today

Thank you very much. No rush. I feel humbled, because I am pretty sure, I am misunderstanding the steps, but maybe this issue will help some other beginner too.

Have a nice day, Tomas

ramilmsh, 2021-01-18:
@TomasHricina you had a few minor problems

1. The tree file should be accessible to the compiled program. For instance, if you wanna load "./tree.xml", after you compile, your hello_BT executable should be in the same folder as your tree.xml. In production, I would advise to create a separate space, where the tree.xml will be kept, so that it is accessible from anywhere. For now: just copy your tree file to wherever you compile the program. (if you plan on using ROS, there are predefined ways to do this)
2. dummy_nodes.h only contains definitions, but not the implementations of the functions, so you need dummy_nodes.cpp as well (don't forget to add it to the CMakeLists.txt), but
3. You don't need to do "using namespace DummyNodes" or dummy_nodes.h for that matter, since you have defined your own ApproachObject
4. I left both dummy_nodes.h and dummy_nodes.cpp, in case you wanna use them in the future, but for now: it compiles just fine and executes your ApproachObject class (i would rename it, to avoid confusion).
[image]

If you have any further questions, feel free to ask

TomasHricina, 2021-01-18:
Thank you, for your time and the pull request @ramilmsh 

It did not work for me, but it must be because I am doing something wrong.
I will list the steps, I have done, so maybe you can spot the problem.

1. git clone @facontidavide s repository, main branch
2. cd BehaviorTree.CPP
3.  `sudo apt-get install libzmq3-dev libboost-dev`
```
 mkdir build; cd build
 cmake ..
 make
 sudo make install
```
4. cd .. (into BehaviourTree.CPP folder)
5. git clone repo you have fixed, cd into it
6. mkdir build, cd build
7. cmake ..
```
Error:

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
CMake Error at CMakeLists.txt:8 (add_executable):
  Target "hello_BT" links to target "Boost::coroutine" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
``
```

And if I try cmake --build . 
```
[ 33%] Building CXX object CMakeFiles/hello_BT.dir/dummy_nodes.cpp.o
[ 66%] Building CXX object CMakeFiles/hello_BT.dir/t01_build_your_first_tree.cpp.o
/home/oooo/mbt/t01_build_your_first_tree.cpp: In function ‘int main()’:
/home/oooo/mbt/t01_build_your_first_tree.cpp:29:10: error: ‘struct BT::Tree’ has no member named ‘tickRoot’
   29 |     tree.tickRoot();
      |          ^~~~~~~~
make[2]: *** [CMakeFiles/hello_BT.dir/build.make:96: CMakeFiles/hello_BT.dir/t01_build_your_first_tree.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:96: CMakeFiles/hello_BT.dir/all] Error 2
make: *** [Makefile:104: all] Error 2

```
Since the error mentions "Boost::coroutine" - I tried same steps with branch called "boost_coroutine", with no luck. That is probably just coincidence.

ramilmsh, 2021-01-19:
@TomasHricina after this, I am not sure how I could help you further, but I have setup a fresh ubuntu:xenial docker container, meaning it was empty in the extreme. I have then done this:

    1  apt update
    2  apt install git
    3  cmake
    4  sudo apt install cmake
    5  apt install cmake
    6  git clone https://github.com/BehaviorTree/BehaviorTree.CPP.git
    7  cd BehaviorTree.CPP/
    8  ls
    9  mkdir build
   10  cd build/
   11  apt-get install libzmq3-dev libboost-dev
   12  cmake ..
   13  sudo apt-get install build-essential
   14  apt-get install build-essential
   15  cmake ..
   16  apt install libgtest-dev
   17  cmake ..
   18  cmake .. -DBUILD_UNIT_TESTS=0
   19  make -j8
   20  make install
   21  cd ..
   22  ls
   23  ldconfig
   24  git clone https://github.com/TomasHricina/BehaviorTrees.git
   25  cd BehaviorTrees/
   26  mkdir build
   27  cd build/
   28  cmake ..
   29  make -j8
   30  ls
   31  cp ../my_tree.xml .
   32  ./hello_BT 
   33  history

> You login as root by default in docker, which is why when I sometimes accidentally type sudo as a habit, I then follow up with the correct version

These are all the commands I typed after starting a fresh, stripped-down installation of ubuntu. The code compiled (though, I did not bother fixing the gtest issue), the code with my modifications worked.

[image]

So, possibly, your ubuntu installation is broken (which happens accidentally all the time), so you could try either docker or running it on LiveUSB. At any rate, it should work

TomasHricina, 2021-01-19:
> @TomasHricina after this, I am not sure how I could help you further, but I have setup a fresh ubuntu:xenial docker container, meaning it was empty in the extreme. I have then done this:
> So, possibly, your ubuntu installation is broken (which happens accidentally all the time), so you could try either docker or running it on LiveUSB. At any rate, it should work

Thank you very much @ramilmsh, you have really done maximum that you could.
I follow your steps exactly and I end up with same error as before.
It looks like there is problem with boost::coroutine.

I will try the docker, you have provided, but in the end, I cannot run in docker, so I have to solve this. I will also check in VirtualBox with fresh Ubuntu and report back.

For completness, here is the CMAKE error, when doing the final "cmake .."
`-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
CMake Error at CMakeLists.txt:9 (add_executable):
  Target "hello_BT" links to target "Boost::coroutine" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?

`
After checking boost version it is: Version: 1.71.0.0ubuntu2

ramilmsh, 2021-01-20:
> d I end up with same error as before.
> It looks like there is problem
I mean, you could try:
`sudo apt install libboost-all-dev`
not sure why it would work, but it might

TomasHricina, 2021-01-20:
> > d I end up with same error as before.
> > It looks like there is problem
> 
> I mean, you could try:
> `sudo apt install libboost-all-dev`
> not sure why it would work, but it might

Already installed. I will try all steps on fresh VM today

TomasHricina, 2021-01-20:
Hello, once again. I have tried the steps @ramilmsh provided on freshly installed and updated Ubuntu 20.04 AMD64 - in Virtual box. Should I open, new issue or keep it here @facontidavide ? 

Here are the steps, with final Cmake error:
```
    1  sudo apt update
    3  sudo apt install git
    4  cmake
    6  sudo apt install cmake
    7  git clone https://github.com/BehaviorTree/BehaviorTree.CPP.git
    8  cd BehaviorTree.CPP/
    9  mkdir build
   10  cd build/
   11  sudo apt-get install libzmq3-dev libboost-dev
   12  cmake ..
   13  sudo apt-get install build-essential
   14  cmake ..
   15  sudo apt install libgtest-dev
   16  sudo apt install libboost-all-dev
   17  cmake ..
   18  make -j8
   19  l
   20  ..
   21  cd ..
   22  ldconfig
   23  sudo ldconfig
   24  git clone https://github.com/TomasHricina/BehaviorTrees.git
   25  cd BehaviorTrees/
   26  mkdir build
   27  cd build
   28  cmake ..
   33  cd ~
   34  history > report.txt

CMake Warning at CMakeLists.txt:7 (find_package):
  By not providing "FindBehaviorTreeV3.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "BehaviorTreeV3", but CMake did not find one.

  Could not find a package configuration file provided by "BehaviorTreeV3"
  with any of the following names:

    BehaviorTreeV3Config.cmake
    behaviortreev3-config.cmake

  Add the installation prefix of "BehaviorTreeV3" to CMAKE_PREFIX_PATH or set
  "BehaviorTreeV3_DIR" to a directory containing one of the above files.  If
  "BehaviorTreeV3" provides a separate development package or SDK, be sure it
  has been installed.

CMake Error at CMakeLists.txt:9 (add_executable):
  Target "hello_BT" links to target "BT::behaviortree_cpp_v3" but the target
  was not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

CMake Generate step failed.  Build files cannot be regenerated correctly.
```
Thank you for help, have a nice day 
Tomas

facontidavide, 2021-01-20:
Let me have a look at it

facontidavide, 2021-01-20:
just add this line to your cmake ant it will work:

find_package(Boost COMPONENTS coroutine QUIET)

closed 2021-01-20 by facontidavide

facontidavide, 2021-01-20:
I just pushed a change that will prevent your problem in the future

dt1729, 2023-01-27:
Hi davide, I'm having this error: 

```
Errors     << planning:check /home/dt/HRIPapers/BehaviorTreeWork/logs/planning/build.check.029.log                                                                   
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "behaviortree_cpp_v3" with any of the following names:

    behaviortree_cpp_v3Config.cmake
    behaviortree_cpp_v3-config.cmake

  Add the installation prefix of "behaviortree_cpp_v3" to CMAKE_PREFIX_PATH
  or set "behaviortree_cpp_v3_DIR" to a directory containing one of the above
  files.  If "behaviortree_cpp_v3" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:13 (find_package)
```

however i have those files located in:

```
usr/local/lib/cmake/behaviortree_cpp_v3
```

here's what my CMakeLists.txt looks like, can you help me with this?

```
cmake_minimum_required(VERSION 3.0.2)
project(planning)

add_definitions(-std=c++11)

## Compile as C++11, supported in ROS Kinetic and newer
# add_compile_options(-std=c++11)

## Find catkin macros and libraries
## if COMPONENTS list like find_package(catkin REQUIRED COMPONENTS xyz)
## is used, also find other catkin packages

find_package(Boost COMPONENTS coroutine QUIET)

find_package(catkin REQUIRED COMPONENTS
  roscpp 
  behaviortree_cpp_v3
  moveit_core
  moveit_visual_tools
  moveit_ros_planning
  moveit_ros_planning_interface
)

## System dependencies are found with CMake's conventions
# find_package(Boost REQUIRED COMPONENTS coroutine system)

################################################
## Declare ROS messages, services and actions ##
################################################

## To declare and build messages, services or actions from within this
## package, follow these steps:
## * Let MSG_DEP_SET be the set of packages whose message types you use in
##   your messages/services/actions (e.g. std_msgs, actionlib_msgs, ...).
## * In the file package.xml:
##   * add a build_depend tag for "message_generation"
##   * add a build_depend and a exec_depend tag for each package in MSG_DEP_SET
##   * If MSG_DEP_SET isn't empty the following dependency has been pulled in
##     but can be declared for certainty nonetheless:
##     * add a exec_depend tag for "message_runtime"
## * In this file (CMakeLists.txt):
##   * add "message_generation" and every package in MSG_DEP_SET to
##     find_package(catkin REQUIRED COMPONENTS ...)
##   * add "message_runtime" and every package in MSG_DEP_SET to
##     catkin_package(CATKIN_DEPENDS ...)
##   * uncomment the add_*_files sections below as needed
##     and list every .msg/.srv/.action file to be processed
##   * uncomment the generate_messages entry below
##   * add every package in MSG_DEP_SET to generate_messages(DEPENDENCIES ...)

## Generate added messages and services with any dependencies listed here
# generate_messages(
#   DEPENDENCIES
#   std_msgs  # Or other packages containing msgs
# )

################################################
## Declare ROS dynamic reconfigure parameters ##
################################################

## To declare and build dynamic reconfigure parameters within this
## package, follow these steps:
## * In the file package.xml:
##   * add a build_depend and a exec_depend tag for "dynamic_reconfigure"
## * In this file (CMakeLists.txt):
##   * add "dynamic_reconfigure" to
##     find_package(catkin REQUIRED COMPONENTS ...)
##   * uncomment the "generate_dynamic_reconfigure_options" section below
##     and list every .cfg file to be processed

## Generate dynamic reconfigure parameters in the 'cfg' folder
# generate_dynamic_reconfigure_options(
#   cfg/DynReconf1.cfg
#   cfg/DynReconf2.cfg
# )

###################################
## catkin specific configuration ##
###################################
## The catkin_package macro generates cmake config files for your package
## Declare things to be passed to dependent projects
## INCLUDE_DIRS: uncomment this if your package contains header files
## LIBRARIES: libraries you create in this project that dependent projects also need
## CATKIN_DEPENDS: catkin_packages dependent projects also need
## DEPENDS: system dependencies of this project that dependent projects also need

catkin_package(
  INCLUDE_DIRS include
  CATKIN_DEPENDS roscpp
  moveit_core
  moveit_visual_tools
  moveit_ros_planning_interface
)

###########
## Build ##
###########

## Specify additional locations of header files
## Your package locations should be listed before other locations
## Declare a C++ library
add_library(planning SHARED src/planning.cpp include/planning.h)

include_directories(
  include
  ${catkin_INCLUDE_DIRS}
)

## Add cmake target dependencies of the library
## as an example, code may need to be generated before libraries
## either from message generation or dynamic reconfigure
# add_dependencies(${PROJECT_NAME} ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})

## Declare a C++ executable
## With catkin_make all packages are built within a single CMake context
## The recommended prefix ensures that target names across packages don't collide

add_executable(run src/run.cpp)
target_link_libraries(run ${catkin_LIBRARIES} planning)

## Rename C++ executable without prefix
## The above recommended prefix causes long target names, the following renames the
## target back to the shorter version for ease of user use
## e.g. "rosrun someones_pkg node" instead of "rosrun someones_pkg someones_pkg_node"
# set_target_properties(${PROJECT_NAME}_node PROPERTIES OUTPUT_NAME node PREFIX "")

## Add cmake target dependencies of the executable
## same as for the library above
# add_dependencies(${PROJECT_NAME}_node ${${PROJECT_NAME}_EXPORTED_TARGETS} ${catkin_EXPORTED_TARGETS})
```

dt1729, 2023-01-27:
Okay I when i did ``` sudo apt-get install ros-<ros_distro>-behaviortree-cpp-v3 ``` it worked however my source make install one didn't weird.