BehaviorTree/BehaviorTree.CPP #252: Injecting an object into all nodes

sjpritchard, 2021-01-20:
Is there a way to inject an object into all nodes at once? The Additional Arguments documentation describes a way to pass arguments on a per-node-type basis, but not (that I can see) to pass an argument for all nodes, regardless of type.

facontidavide, 2021-01-20:
no, it is not possible

closed 2021-01-20 by facontidavide

ramilmsh, 2021-01-21:
> Is there a way to inject an object into all nodes at once? The Additional Arguments documentation describes a way to pass arguments on a per-node-type basis, but not (that I can see) to pass an argument for all nodes, regardless of type.

If all of your nodes are of the same type (say, an ActionNode), you could create a class on top of it, and then inherit your class instead of the ActionNode