BehaviorTree/BehaviorTree.CPP #253: Release into ROS 2 Rolling

clalancette, 2021-01-26:
It looks like this package is currently failing to build in ROS 2 Rolling: https://build.ros2.org/view/Rbin_uF64/job/Rbin_uF64__behaviortree_cpp_v3__ubuntu_focal_amd64__binary/ .  The problem seems to be a missing `test_depend` on `ament_cmake_gtest`.

The good news is that this looks like it was already fixed on the master branch in commit a331b211eabc4ae1b82f112192be43ebf58b067b .  Could you do a `bloom-release` of the latest into Rolling so we can get that fix there?  Thanks.

facontidavide, 2021-01-31:
For some reason my attempt to do bloom_release fails:

```
Your track's 'actions' configuration is not the same as the default.
Unless you have manually modified your 'actions' (the commands which get run for a release), you should update to the new default.
Should it be updated to the default setting?
Continue [Y/n]? 
==> Testing for push permission on release repository
==> git remote -v
origin	https://github.com/ros2-gbp/behaviortree_cpp-release.git (fetch)
origin	https://github.com/ros2-gbp/behaviortree_cpp-release.git (push)
==> git push --dry-run
Username for 'https://github.com': facontidavide
Password for 'https://facontidavide@github.com': 
remote: Permission to ros2-gbp/behaviortree_cpp-release.git denied to facontidavide.
fatal: unable to access 'https://github.com/ros2-gbp/behaviortree_cpp-release.git/': The requested URL returned error: 403
Cannot push to remote release repository.
Hint: If you just typed in your username/password and you have two-factor authentication,see:
  http://wiki.ros.org/bloom/Tutorials/GithubManualAuthorization

```

clalancette, 2021-01-31:
I've just added you as a maintainer on https://github.com/ros2-gbp/behaviortree_cpp-release.  Please try again.

facontidavide, 2021-02-03:
I think it is done

closed 2021-02-03 by facontidavide

clalancette, 2021-02-04:
Yep, thank you!