BehaviorTree/BehaviorTree.CPP #258: apt-get installs version 3.1.1

asasine, 2021-02-05:
Hello!

When I installed the package through apt-get, it appears to have installed version 3.1.1. This caused some trouble when following the tutorials as the `tree.tickRoot()` method is not available in this version of the library.

```
adam@ADAM:~$ sudo apt-get install ros-dashing-behaviortree-cpp-v3
adam@ADAM:~$ cat /opt/ros/dashing/share/behaviortree_cpp_v3/package.xml
<package>
  <name>behaviortree_cpp_v3</name>
  <version>3.1.1</version>
  <description>
  This package provides the Behavior Trees core library.
  </description>

  <maintainer email="davide.faconti@gmail.com">Davide Faconti</maintainer>

  <license>MIT</license>

  <author>Michele Colledanchise</author>
  <author>Davide Faconti</author>

  <test_depend>ament_cmake_gtest</test_depend>
  <buildtool_depend>ament_cmake</buildtool_depend>
  <export>
     <build_type>ament_cmake</build_type>
  </export>

  <build_depend>libzmq3-dev</build_depend>
  <run_depend>libzmq3-dev</run_depend>

</package>
```

The _package.xml_ shows the version as 3.1.1 in the third line.

How can I get the latest version of the library using apt-get and/or rosdep?

---
Other info
```
adam@ADAM-DEV:~$ printenv | grep ROS
ROS_MASTER_URI=http://localhost:11311
ROS_VERSION=2
ROS_PYTHON_VERSION=3
ROS_DISTRO=dashing
adam@ADAM-DEV:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.5 LTS
Release:        18.04
Codename:       bionic
```

facontidavide, 2021-02-05:
I am not maintaining version older than foxy

asasine, 2021-02-05:
Thanks for the quick reply! We aren't able to upgrade our system to Ubuntu 20.04 at this time so Foxy is out of reach for us. If we went with compiling locally instead of installing through apt-get/rosdep, do you know of any blockers with compiling on Ubuntu 18.04?

facontidavide, 2021-02-05:
it should work just fine.

closed 2021-02-05 by facontidavide