BehaviorTree/BehaviorTree.CPP #263: SequenceNode ticking at most 1 Action per tree.tickRoot()

ekouters, 2021-03-04:
I am looking for a SequenceNode ticking at most 1 Action during a `tree.tickRoot()`.
My question: Is a custom SequenceNode the only solution?

**Example**
Suppose the following tree:

```
<SequenceStar>
   <ActionA/>
   <ActionB/>
</SequenceStar>
```

- The first few calls to `tree.tickRoot()` will tick ActionA, returning RUNNING.
- At some point, `tree.tickRoot()` will tick ActionA, returning SUCCESS.
- During the same `tree.tickRoot()` as ActionA returned SUCCESS, ActionB is ticked.
- In other words, during one call to `tree.tickRoot()`, two Actions were ticked.

This is because SequenceStarNode does not do a `return` on SUCCESS, but continues the `while` loop instead:
https://github.com/BehaviorTree/BehaviorTree.CPP/blob/master/src/controls/sequence_star_node.cpp#L53

facontidavide, 2021-03-04:
I am going to blow your mind.

You can create and register custom control nodes exactly as you do it with action nodes. 

Build your own

closed 2021-03-04 by facontidavide

ekouters, 2021-03-04:
> My question: Is a custom SequenceNode the only solution?

I'm going to assume your answer to this question was "Yes".