BehaviorTree/BehaviorTree.CPP #265: Resetting the tree

meghdeepj, 2021-03-05:
I am BT.CPP-v3 on ROS Melodic

I need to reset the complete behavior tree ( or a subtree) and tick the root's children from the beginning when I receive the relevant feedback from my robot's GUI. What is the correct and clean procedure for that?
I have tried breaking out of the tick loop, calling the tree object destructor and creating the object again from xml file. But the ROS Node crashes.
Thanks for your help.

facontidavide, 2021-03-05:
I would say that this is an antipattern. Basically you are trying to bypass the logic of the tree.

I understand that could be conveninient, but defintively a "code smell".

On the other hand, it would be interesting to know why it is crashing. Can you give me an example to reproduce the issue?

meghdeepj, 2021-03-05:
I was trying to get this quick hack going until I add the reset logic to my Tree, I guess I will go ahead with the proper way. I tried this code:
```
bool reset=false;

void resetCb(const std_msgs::UInt8MultiArray::ConstPtr& msg){
  if(msg->data[1]==1){
    reset=true;
  }else{
    reset=false;
  }
}

int main(int argc, char **argv)
{
  float tick_rate;
  ros::init(argc, argv, "bt_navigator");
  ros::NodeHandle nh;

  ros::Subscriber reset_sub = nh.subscribe<std_msgs::UInt8MultiArray>("from_navigation",10, resetCb);
  ros::Publisher stat_pub = nh.advertise<std_msgs::Int8>("bt_status", 1);
  ros::param::get("/bt_navigator/bt_tick_rate", tick_rate);
  ros::Rate loop_rate(10);
  
  while(ros::ok()){    
    if(!reset){
      BehaviorTreeFactory factory;
      // register rosservice client action nodes
      factory.registerNodeType<peppermint_bt::HasPathCondition>("HasPathCondition");
      factory.registerNodeType<peppermint_bt::IsQueueEmptyCondition>("IsQueueEmptyCondition");  
      factory.registerNodeType<peppermint_bt::IsReplanCondition>("IsReplanCondition");  
      // load tree from xml (file/text)
      auto tree = factory.createTreeFromFile("./path_tree.xml");
      // set initial status
      NodeStatus status = NodeStatus::IDLE;
      // initialize loggers (terminal and groot)
      StdCoutLogger logger_cout(tree);
      PublisherZMQ publisher_zmq(tree);
      // sleep time for subscribers and timers to initialise callbacks
      ros::Duration init_time(2.0);
      init_time.sleep();
      stat_pub.publish(3);
      ros::spinOnce();
      // tick the root at rate Hz until ros is shutdown or tree returns FAILURE or SUCCESS
      ros::Rate rate(tick_rate);
      while( ros::ok() && (status == NodeStatus::IDLE || status == NodeStatus::RUNNING))
      {
        status = tree.tickRoot();
        if(status==NodeStatus::RUNNING){
          stat_pub.publish(2);
        }else{
          stat_pub.publish(3);
        }
        rate.sleep();
        ros::spinOnce();
        
        if(reset==true){
          break;
        }
      }
    tree.~Tree();
    }
    loop_rate.sleep();
  }
  
  return 0;
}
```

closed 2021-04-01 by facontidavide